portapack-mayhem/mayhem-firmware #515: Does not detect Tesla TPMS

jLynx, 2022-03-24:
**Describe the bug**
For some reason the TPMS app does not seem to detect the Tesla TPMS sensors. You are able to view the telsa screen populate the data, yet the HackRF does not detect them. 

I am wondering if this is because it's using a slightly different format that is not added into the TPMS app yet. I'll also do some digging. 

**Expected behavior**
The sensors should show up like all other brands

**Affected versions**
All versions including the current 

**Additional**
You can find the FCC filing for the TPMS sensor here https://fccid.io/2AEIM-1472547

jLynx, 2022-03-24:
It's also possible that these sensors use BLE and not 315/433 Mhz

jLynx, 2022-03-24:
[image]
So after looking at the FCC filings, I can see it does indeed communicate over Bluetooth LE. I will see if I can do some sniffing around and see how this information is produced

jLynx, 2022-04-03:
Here are some screenshots I was able to get from my Tesla looking at the TPMS data 

[image]
[image]
[image]
[image]
[image]

eried, 2022-04-03:
ok so actually it does not really belong to the current tpms app, that only works with non BLE sensors

jLynx, 2022-04-03:
Correct, but I would not close the issue. Let's change it to a feature request.