portapack-mayhem/mayhem-firmware #518: Bootstrap source code missing, dead link in wiki

CONIGUERO, 2022-03-26:
While reading the Wiki page on the firmware boot process, I saw it mentions the bootstrap code that runs on the M4 cortex as one of the initial stages.

However when clicking the [hyperlink](https://github.com/sharebrained/portapack-hackrf/blob/master/firmware/bootstrap/bootstrap.c) I see that it's returning a 404. Manually browsing the repository I discover there's no bootstrap folder.

notpike, 2022-03-28:
Hi,

This Wiki was originally from the [sharebrained/portapack-hackrf wiki](https://github.com/sharebrained/portapack-hackrf/wiki/Boot-Process) and as you mentioned out of date. I'll see if I can track down the current method of how this bootstraps and update the wiki. :D

CONIGUERO, 2022-03-29:
> Hi,
> 
> This Wiki was originally from the [sharebrained/portapack-hackrf wiki](https://github.com/sharebrained/portapack-hackrf/wiki/Boot-Process) and as you mentioned out of date. I'll see if I can track down the current method of how this bootstraps and update the wiki. :D

Ah, nice.

Also, in this actual fork repo, there *is* a bootstrap folder, albeit only containing binaries.

Where is the source code for "bootstrap.bin" and "bootatrap_pad.bin"?

notpike, 2022-03-29:
> Ah, nice.
>
> Also, in this actual fork repo, there _is_ a bootstrap folder, albeit only containing binaries.
>
> Where is the source code for "bootstrap.bin" and "bootatrap_pad.bin"?

Hi,

I think the bootstrap folder is just a left over thing that hasn't been removed. Back in 2019 bootstrap was removed from the original portapack code base and isn't used any more. I'm not sure if the current wiki reflects the current method or not.

https://github.com/sharebrained/portapack-hackrf/commit/be61b32efe477f4b075c6e6b632f37ea94f9c3de

ArjanOnwezen, 2022-04-06:
Hyperlink is removed

closed 2022-04-06 by ArjanOnwezen