portapack-mayhem/mayhem-firmware #522: CW initial low power

100monkeys, 2022-03-29:
**Describe the bug**
A clear and concise description of what the bug is. 

CW in "cw" modulation starts with very low gain.
After switching to "fm" modulation and then back to "cw" modulation then rf output level is normal,

A few times it keys the transmitter but at same time it frequency modulates the signal with a lower audio buzz tone.
(I can not reproduce that last problem as this seem to occur randomly )

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Transmit"
2.  select  "Morse"
3. set frequency
4. default modulation is "CW"
5. check rf level
6. change modulation to "fm" and back to "cw"
7. check rf level.

**Expected behavior**
Sending the word  "portapack" in morse continuous wave at the default gain level of 47

**Affected versions**
1.4.3

notpike, 2022-03-29:
Which portapack are you using? 

If it's the H2 with the battery did it make a difference while it was plugged in or not?

Thanks!

gullradriel, 2023-04-26:
@100monkeys do you still have the problem with latest nightly ?

PR #1629 (2023-12-08): Fix Morse CW mode transmit
Fixes #522

With this change, CW transmissions are heard without needing to transmit in FM mode first.