portapack-mayhem/mayhem-firmware #524: [addon feature maybe?]Car Charging Port Opener

Darkcast, 2022-03-29:
found this and I though it would be nice to add it to the firmware 
https://github.com/jimilinuxguy/Tesla-Charging-Port-Opener

notpike, 2022-03-29:
> found this and I though it would be nice to add it to the firmware https://github.com/jimilinuxguy/Tesla-Charging-Port-Opener

Hi,

The Tesla charging port capture can be found in the mayhem_v1.4.4_COPY_TO_SDCARD.7z zip. This download is available at 
https://github.com/eried/portapack-mayhem/releases. The capture of the Tesla charging port message can be found under SAMPLES/TeslaChargePort_EU_AUS.C16. 

If you're trying to use this in the US you'll need to change the frequency on the text file TeslaChargePort_EU_AUS.TXT to `center_frequency=315000000`.

Thanks!

ArjanOnwezen, 2022-03-29:
See https://github.com/eried/portapack-mayhem/blob/next/sdcard/SAMPLES/TeslaChargePort_EU_AUS.C16

closed 2022-03-29 by ArjanOnwezen

eried, 2022-03-30:
> > found this and I though it would be nice to add it to the firmware https://github.com/jimilinuxguy/Tesla-Charging-Port-Opener
> 
> Hi,
> 
> The Tesla charging port capture is apart of the latest release and can be found in the [mayhem_v1.4.4_COPY_TO_SDCARD.7z](https://github.com/eried/portapack-mayhem/releases/download/v1.4.4/mayhem_v1.4.4_COPY_TO_SDCARD.7z) zip. This download is available at https://github.com/eried/portapack-mayhem/releases. The capture of the Tesla charging port message can be found under SAMPLES/TeslaChargePort_EU_AUS.C16.
> 
> If you're trying to use this in the US you'll need to change the frequency on the text file TeslaChargePort_EU_AUS.TXT to `center_frequency=315000000`.
> 
> Thanks!

is it only that for US? :o

ArjanOnwezen, 2022-03-30:
@eried Perhaps good to add those files as well and rename them to TeslaChargePort_US.C16

eried, 2022-03-30:
yes I agree, but @jLynx had a better capture file

ArjanOnwezen, 2022-03-30:
Please continue comments on #229