portapack-mayhem/mayhem-firmware #525: Problem upgrading the firmware

Professa-Dementia, 2022-03-30:
History:
I have an original HackRF One, not a clone.  I also have an original / official Portapack H1

It has been a while since I used the device.  The last time, I upgraded it to the latest firmware at the time, which was 1.3.0.  This went without any problems, as I recall.

Today, I tried to upgrade to the latest, 1.4.4

When I do the upgrade and reboot, the screen turns on white briefly, then turns black and stays black.

Troubleshooting:
The USB cable is fine and works reliably.

I figured out that the device is actually working correctly, except for the screen.  I can use the touch screen to select HackRF mode in the lower right corner, then click the scroll wheel to select YES (to confirm HackRF mode).  Windows then makes the "new device detected" sound, and recognizes the HackRF.
Some firmware versions show the splash screen, so I might have to click the scroll wheel right after restarting, to get past this screen.

NOTE: during all the above, the screen is blank, I am just navigating from memory.

I tried upgrading and downgrading to the various versions.  What I found was that 1.3.1 and below works and the screen works properly.  If I upgrade to 1.4.0 or above, upon restart, the screen flashes white briefly, then turns black and stays black.  If I then go into HackRF mode and downgrade to 1.3.1, the screen starts working again.

It appears that between 1.3.1 and 1.4.0 something changed in the software that is causing problems with the screen, or maybe just the backlight.  To see if it was just a backlight issue, I shone a bright flashlight at the black screen.  On many LCDs, you can often see the image on the display this way, even when the backlight is off.  However, when I shone the flashilight on the screen, I did not see any image.

SUMMARY:
1) Upgrading to 1.4.0 or higher results in a black screen, although the device seems to be functioning normally, otherwise.
2) I have an original Great Scott HackRF One and official ShareBrained PortaPack H1 - not clones or knock offs.
3) My USB cable appears to be working properly
4) The upgrade / downgrade procedure by putting the device in HackRF mode seems to be working properly.
5) It appears a change between Mayhem v1.3.1 and v1.4.0 is causing this problem.
6) I have gone through other reports of people with similar issues, but found nothing that has helped so far.

Any suggestions are appreciated.  Is my HackRF or PortaPack H1 just too old and not supported under the current firmware?

ArjanOnwezen, 2022-03-30:
Duplicate of #346

closed 2022-03-30 by ArjanOnwezen

sivaelid, 2022-03-30:
Program PortaPack on Windows Machine.                SIVAELID   11/03/22

THis is my approach.

1.    Use a good USB lead.
2.    Set portapack in to DFU mode by holding both DFU and RESET buttons and then release the RESET, the one nearest to case edge.
3.    The LEDs will now show 3v3, RF at full brightness the others a low level.
4.    Using latest version firmware compiled ZIP (1.4.4) uncompressing it in a file location. From this now select “dfu_hackrf_one.bat” and run this following the prompts. The PortaPack will now show 3v3,1v8, RF lights full brightness and the rest flashing.
5.    Now select from “flash_hackrf_one.bat”, run this and follow the prompts. The PortaPack will now need to be restarted. It will now return after a longish delay the splash screen, and should now be working when selected.
6.    If this does not work then need to clear device manager first before enabling driver see below. 
If standard install does not work as is often the case for me and it will not “Find the HackRF” showing as 0 HackRF the follow this step by step. 
1.    Check that HackRF/ Portapack is not connected.
2.    Go to Device Manager on your windows PC, and at the top View> Select Hidden Devices and select.
3.    Go to “Other device’s” and see if LPC is shown If this is this is Ok.
4.    Go to “Universal Serial Bus devices, at the bottom of the list and open it will now show and number of device and probably include “LPC” and “HackRF One” sometimes more than one copy. Delete all of the LPC and HackRF one items so none are left. Check again the list of devices go to Item 2 and 4 above.
5.    Connect via USB using a good lead to the HackRF / PortaPack to the Windows PC
6.    Use a good USB lead.
7.    Set portapack in to DFU mode by holding both DFU and RESET buttons and then release the RESET, the one nearest to case edge.
8.    Now using Zadig Driver https://zadig.akeo.ie/. follow the instruction install it if not already installed and then, select option> List all device’s, you should now see either LPC or HackRF. Select the items and update using the recommended driver.
9.    Now revert to the standard Install at the top of the page, it should now work.