portapack-mayhem/mayhem-firmware #545: No Audio on H2 since 1.5.0 / nightly 2022-04-06

xstff78, 2022-04-07:
Hi,

On my H2, there is no audio since the nightly 2022-04-06, which worked fine on previous releases.

- Interface – deselect “Hide H1 Speaker Option“ (red cross)
- Activate speaker (green symbol in title bar)
- Receiver – Audio –> no sound

I also noticed some display flickering and changes in the bottom bar (disappear and reappear of release, date, time) after changing the speaker state in the top bar following by pressing the down button – maybe some state memory locations are mixed up?

Kind regards,
Christian

jLynx, 2022-04-07:
Are you able to share a picture of the back side of your portapack? I need to see what audio chip it has

xstff78, 2022-04-07:
Thank you for your quick reply. The H2 is labelled with "PCB v3.4b"; please find a picture below.

[image]

jLynx, 2022-04-07:
Can you please go to debug->Peripherals and then take a photo?

xstff78, 2022-04-07:
I have just checked again; audio works up to (and including) nightly 2022-04-05.
Please find the screenshots (1.5.0) below.

[image] [image]

sck-nogas, 2022-04-07:
This has been fixed in this update for my H1. 
https://github.com/eried/portapack-mayhem/pull/547#issuecomment-1091429701

jLynx, 2022-04-07:
That's odd. Nothing has changed. But if it's Woking I'll close this issue. Please re open if it happens again

xstff78, 2022-04-08:
For my device (see above), 59d08a4 or nightly 220408 could not fix the problem - still no working audio. Hardware seems to be ok; audio works with n 220405.

jLynx, 2022-04-08:
Oh sorry I misunderstood. I thought you said it was working in  1.5.0. I'll re open the issue

jLynx, 2022-04-08:
Does audio not work with the speaker, headphone jack or both?

xstff78, 2022-04-08:
Just tested - working / not working applies to both, speaker and headphone jack.

jLynx, 2022-04-08:
Add me on discord and I'll send you a custom firmware to try jLynx#7987

godaminski, 2022-04-10:
I am having the same issue. I do not have audio from either the internal speaker or headphone jack since 220408 update.

godaminski, 2022-04-10:
I can confirm that the audio issue is corrected in the 20220409 nightly build.

sivaelid, 2022-04-10:
The "no audio issue"  is not corrected on a H2+ with the nightly build 20220409 or other version from v1.5.0. The last working version was v1.4.4.

ArjanOnwezen, 2022-04-10:
@xstff78 could you report back whether 20220409 nightly is working or not?

GullCode, 2022-04-10:
Have you tried one of the latest nightly ? Can you try with one of the latest bin file in test drive ?

xstff78, 2022-04-10:
@ArjanOnwezen Unfortunately no success for my hardware version - no audio with n 220409.

jLynx, 2022-04-10:
I am back today so I can work on a fix. @xstff78 message me on Discord when you are online so we can get started

vk4khp2, 2022-04-10:
Same with my H2+ on N 220409, no Audio on either speaker or headphone socket.
the other strange issue, is at start up, after the update, I have to hold down the Up button each time i start the device.
or it's just a black screen.

godaminski, 2022-04-11:
So here's an interesting one. The holdup battery in my PortaPack was dead. I just replaced it, which corrected another issue with needing to press the left button when powering on, but now I'm back to no audio.

jLynx, 2022-04-11:
@godaminski please send me a message on Discord and I will send you a custom firmware to test jLynx#7987

jLynx, 2022-04-11:
We will keep this issue open till the PR has been merged in

godaminski, 2022-04-11:
> @godaminski please send me a message on Discord and I will send you a custom firmware to test jLynx#7987

I am not up on Discord but will try to get up on it.

closed 2022-04-11 by jLynx