portapack-mayhem/mayhem-firmware #552: Save settings on Portapacks without battery

doccodyblue, 2022-04-08:
On my PP I have to reenter all settings after every powercycle.

I don't have a backup battery attached right now because I simply couldn't find one that fits into the chinese battery holder.

Wouldn't it be nice to save settings to the sdcard in addition? I will add a battery eventually but I assume there are others around like me.

Just an idea.

Best, Andy

jLynx, 2022-04-08:
Yes that is something that we have talked about doing and will look into doing that at a later date

doccodyblue, 2022-04-08:
Sounds great! Thanks!

ArjanOnwezen, 2022-04-08:
> On my PP I have to reenter all settings after every powercycle.
>
> I don't have a backup battery attached right now because I simply couldn't find one that fits into the chinese battery holder. [...]

Isn't it a CR2032 one?

doccodyblue, 2022-04-08:
Nope, it is way smaller and I have a set that includes nearly every knob battery type but none of them fits in there :( I will try to get a 2032 holder on there

ArjanOnwezen, 2022-04-10:
Else just contact the seller ir return the device.

jLynx, 2022-04-11:
Added this feature in the following PR. Should be merged shortly

jLynx, 2022-04-11:
@doccodyblue could you please test the test firmware here? https://github.com/eried/portapack-mayhem/pull/565#issuecomment-1094512545

doccodyblue, 2022-04-11:
Tested. It doesn't remember the settings. Could you please confirm that FW Version 74d13e6 includes "sd-settings"?
Furthermore I still have to hold "left" on powerup to have the screen working, I hope that doesn't interfere with the load process?
Is there a file I can look for on the SD?

doccodyblue, 2022-04-11:
I did. Observed the following:
have to hold left while power on
/hardware/settings.txt is created and has a "3" in it
No settings are loaded at boot time 

@BH3EMQ correct on the CR1220 -> thanks! 
of cause I removed the battery for all sd-card testings

jLynx, 2022-04-11:
@doccodyblue after that file has been created, can you please hold the center select/OK button on boot and see if then boots up?
Also what device do you have? (H1, H2 etc)

doccodyblue, 2022-04-11:
no boot with center button held and file on the card

Actually the PP is labeled Portapack H2+ HACK RF ONE - The HackRF board is one from Banggood

[image]

jLynx, 2022-04-12:
This has also been fixed in https://github.com/eried/portapack-mayhem/pull/567

closed 2022-04-12 by jLynx