portapack-mayhem/mayhem-firmware #553: No morse in CW mode on frequencies over 2149.999 Mhz

100monkeys, 2022-04-08:
**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Transmit
2. Tap on 'morse'
3.  select frequency 2400 mhz
4. select 'FM'  (this works fine)
5. select 'CW' give just carrier or sometimes a buzzer tone.

**Expected behavior**
'FM' frequency modulated keyed audio tone
'CW' keyed carrier in morse

**Affected versions**
1.4.3     1.4.4

**Additional**
Up to 2149.9999 Mhz things are working fine
from  2150 - 2299 Mhz many tx modes fail
from 2300 onward 'FM' morse works again but CW does not.
the range 2150-2300 is not very interesting but i would like for CW to work on 2400 Mhz uplink
of QO-100 Amateurradio satellite.

Setup:
Original HackRF one with H1 Portapack clone (no battery option)
Tested with AOR SR2000 receiver.

Maybe this has to do with the PP clone  I had some trouble getting it working in tx modes earlier but after a reflash of the cpld i got it working: https://github.com/eried/portapack-mayhem/issues/466#issuecomment-1076686423 )

Brumi-2021, 2022-09-06:
Hi @100monkeys , I think that you solved that issue long time ago . If so , please kindly close that issue .

thanks

100monkeys, 2022-09-06:
Hi Brumi-2021 I could not solve this. Was hoping some one with original portapack had the same. But i guess that is the reason it won't work.
If you want me to i will close it.
Tnx

Brumi-2021, 2022-09-06:
Hi @100monkeys ,
 no no , if it  is NOT solved , do not close it . 

I just thought that reflashing the CPLD you fixed it . 
Now is clear . 
Cheers

gullradriel, 2023-03-31:
It may really be only hardware limitations. HackRF is know to have some best working ranges, and some less good or poor others.

Brumi-2021, 2023-05-21:
Hello @100monkeys , 
I have been checking that issue,  and another one similar that we closed, #708 , 
and you are right , there are really several bugs there, some deterministic , and several randoms.

1 -) Bug easy to reproduce 100% , is that just entering to that Morse App , (if you do not change the MODULATION) the initial default  CW is transmitted  , around -40 or 50 dB's  respect the expected one . ---> that is why you said CW not working . 
[image]

But If you increase TX power to maximum you can still see that TX at -70 dBm , very small ,  close to the noise .
The problem is solved changing the MODULATION , FM and transmit to FM and then  back to CW . and then you got the normal levels.
[image]

2-)but there are other multiple random bugs,  and difficult to reproduce . So far I do not know the sequence to reproduce it . And the sequence that fixed it . But the first test TX in FM and CW  usually works well , but then randomly CW mode is wrongly  transmitting  FM ,
And changing the modulation FM <-->CW , sometimes it is recoverd a good CW transmission .
Those bugs are that sometimes CW is transmitting a continuos FM signal ,  and sometimes morse carrier of the FM signal.
The FM  mode , seems stable and always working well
[image]

100monkeys, 2023-05-21:
Thanks very much for this research. I will follow your suggestions and let you know the results.

gullradriel, 2023-07-18:
@100monkeys any news ?
Have you tested with latest nightly ?

100monkeys, 2023-07-19:
Hi @gullradriel I Just did a quick test with ver 1.7.3 and this problem is still there.

100monkeys, 2023-07-19:
Hi and Thanks for your reaction,

1 -) same results here but only below 2149.999 Mhz. 
When i repeat this test at 2400 mhz  the power increases after mode switch FM and back to CW but the CW does not key but gives a continuous signal. 
It seems related to something that switches after 2149.999 Mhz?  
When i do the mode switch again i get a buzztone in on mode CW. 
This stays the same no matter how often i switch back and forth after this. 
The FM modulated audio CW seems to work fine on all frequencies.
 
So the recover work-around for the CW power works with the CW-FM-CW switch but the constant (unkeyed) carrier in mode CW problem on  2400 mhz remains.
Hope this helps? 
regards 
100monkeys

Brumi-2021, 2023-07-20:
Hi @100monkeys ,
you are right , there are several strange things in that freq band , ...
 
I could reproduce what you said,  but ....

1-)  Regarding  2200-2210 Mhz example, 
 I am not able to transmit /receive  in 2200 or 2210 Mhz example , no way to receive anything in that freq. 
 It does not matter using Mic App TX,  Signal Gen. TX, nor Morse App Tx 
             a-) using Hackrf TX -->   SDR + Hackrf RX .   (no reception )
             b-)  (nor direct Hackrf receiver) .                      (no reception) 
             Not sure , where is the problem , TX or RX side, and if it is HW or SW limittations or just a bug ???

2-) Regarding 2300-2400 Mhz band , 
           a-) I can receive 2300..24000 Mhz Morse TX App ,  using Hackrf TX  Morse App --> SDR + Hackrf RX,  
                        (it works as you said , using Morse App Tx , 
                                FM morse ok,  CW morse NG , just some buzztone mod, not working well the Morse !
                        (but I checked Signal Gen App in those freq. 2300-2400 Mhz and it works well in all type of Signal modulation,
                        CW, Square, sinus, square, triangle , Random noise ....)    

           b-) (but using  Hackrf TX Morse App   ---> direct stand alone , Hackrf Audio App RX , 
             I can not receive 2300 Mhz, (it just starts to receive anything  from a TX set to  2320 Mhz  onwards.  
            It is very strange , using the same Hackrf receiver in a SDR it works at 2300Mhz , 
            but using self Audio App Rx NOT receiving the same carrier  below 2320 Mhz ). 
            Not sure if the problem is in the TX or RX side or new bug in Audio Rx ?

NotherNgineer, 2024-01-27:
PR #1629 might fix the CW issue.
PR #1819 should fix the tuning issue with frequency range 2170-2300 MHz.

Please re-test this issue using nightly n_240128 when it comes out tonight.

Brumi-2021, 2024-01-27:
I will retest it with the new nightly , and most likely we will be able to close that issue . (The ham QO-100 Amateurradio satellite operators will thank you 🙏)
Let’s confirm about it tomorrow . 
Great PR’s Mark !

100monkeys, 2024-01-27:
That would be great! Thanks !

Brumi-2021, 2024-01-28:
Hi @NotherNgineer , I am affraid that we can not close completely that issue .

You solved the non transmission in 2,2Ghz area , 
I confirmed , previous fw ,example 1.91 was doing like that 
[image]

And it is true that you fixed and solved that 2.2Ghz transmission area, now from today's nightly n_240128 , we have nice transmission
[image]

But there are two outstanding problems , 
1-) regarding freq. transmissions, (tested using Mic tx App ) , from  2750 Mhz onwards,  something strange I can not receive those signals from another close hackrf in Hackrf mode with gqrx sdr sw (as example) .
Not sure if Hackrf reception in Hackrf mode is fine from 2,7Ghz to 6 Ghz , (but I am assume that it should be correct ,and the problem is in our mayhem side , but not confirmed) . 
 
2-) If I go to Morse Tx App , even using today's nightly n240128 , it seems that the transmission is always sending FM modulation , even we select CW .   In CW case, I was expecting on-off keying of the carrier without any modulation  , but now ,  I can see in the spectrum FM modulation on top. 

So you fixed the 2,2-2.3 Ghz tx band ,  but we have some other outsanding issues there. 
If you agree,  I  will not close by now that issue , because it is half solved.

zxkmm, 2025-03-19:
>  If I go to Morse Tx App , even using today's nightly n240128 , it seems that the transmission is always sending FM modulation , even we select CW .

I think a part of this issue had been resolved in https://github.com/portapack-mayhem/mayhem-firmware/pull/2436 

And I can't reproduce the issue in this PR anymore.

But I notice another issue, that the pitch of cw will getting higher and higher by transmitting: https://discord.com/channels/719669764804444213/1170738202924044338/1351935635505676401

Speedster04, 2026-03-18:
Thanx for the advice of deleting old apps I didn't recicized that.....