portapack-mayhem/mayhem-firmware #557: Feature Request: Add support for a .gif splash

Cry5talz, 2022-04-09:
Right now we've got a static .bmp splash. I figure it'd be cool to add support for a small .gif splash screen. 
Perhaps in the settings we could have the option to toggle it between looping and non-looping?

schwaaaa, 2022-04-11:
Hi! Saw this issue from Bountysource.

This could help! 

https://stackoverflow.com/questions/52298410/how-to-show-an-animated-gif-in-splash-screen

Kind regards, Schwa

jLynx, 2022-04-12:
Feel free to open a PR @vibhuti019

Allong12, 2022-04-15:
Your main concern with implementing this would be how much space/memory budget you have for frame storage.

A single frame per the [bmp_splash.hpp](https://github.com/eried/portapack-mayhem/blob/f161e85f960cff0c166173f4f7a4244b8c625375/firmware/application/bitmaps/bmp_splash.hpp) file is 3,828 bytes long.

Simplest implementation would turn that array into size [n_frames][3828], with each array being one bitmap frame of the gif, that you've dumped as a bitmap beforehand.

Then at paint time you would just need to replace the reference to  splash_bmp with, splash_bmp[_n_], with an increasing sentinel `n%n_frames`, perhaps incrementing _n_ every overflow of a separate sentinel, which itself is incremented every paint iteration, to help control frame rate

https://github.com/eried/portapack-mayhem/blob/f161e85f960cff0c166173f4f7a4244b8c625375/firmware/application/ui_navigation.cpp#L663

gullradriel, 2023-05-12:
@AlLongley is right, it can be possible but it would be real time and firmware consuming, even if we can load bitmaps from sd.
You can forget the area idea, it's already nice that one can fit in memory.
I'm also questionning the gains. I mean, is it something that serves the device purpose, having a splash screen at boot ?
Well. I am already not using a splash screen, so a gif at start... 
I think that It's not because we can do it that it's a great thing, and that's why I'm closing the issue.
Regards.