portapack-mayhem/mayhem-firmware #559: My encoder direction is reversed and seems to skip options

lu1aax, 2022-04-10:
Rotary encoder direction is reversed (clockwise decreases values and counterclockwise increases values) and moving from one stop point to the next one skips one option or value.

**Affected versions**
All I tried from 1.4.0 on

**Additional**
My H2+ portapack seems to be slighty different physically (eg. I 3D printed a case from thingiverse and screen position is ok but buttons and encoder are in slight different positions) but works ok besides this strange encoder behavior.

ArjanOnwezen, 2022-04-10:
Never heard of this before. Is this the case for all firmware version. If so it looks like a hardware issue and I would urge to return the unit. It's not hard to make a software fix, it it doesn't seem right to add an extra option only for a handful of devices.

closed 2022-04-10 by ArjanOnwezen

lu1aax, 2022-04-10:
Well, it seems like there is another thread describing the same situation. Sure this is hardware related but I don't think this is due to faulty hardware but a different kind of encoder has been used.

ArjanOnwezen, 2022-04-10:
Best to check with your vendor. Or ask for a refund, so you can buy back a compatible one.

lu1aax, 2022-04-10:
I bought this last year, way beyond the chance to return it now. Device is perfectly usable but it's weird behavior makes it hard to. I identify this issue as a duplicate of #444 where other people thinks this shouldn't be closed, and some have found a way to partially solve the problem. I'd like to contribute in solving this in either software by providing more details or hardware if its the case a different class of encoder is installed on my board, i can get a new one and install it to fix this.