portapack-mayhem/mayhem-firmware #562: 1.5.0- Receiver-Audio  has poor/no reception

miotislucifugis, 2022-04-10:
**Describe the bug**
I just flashed 1.5.0 on my H1 portapack and opened Receiver-Audio to do a quick test.    Trying to dial in some local FM stations... I can "see" a signal on the waterfall, but only hearing static.  The static doesnt change much as I scroll up and down the dial.     If i revert the firmware to 1.4.4,  it works perfectly and local stations are crystal clear.  Ive only tried with WFM, in the commercial FM bands... not sure if this bug extends to the other bands/modes... nor have I tried other "receive" apps to see if they are affected.

Amp/lna settings remain the same (default values= amp 0,  32/32)

GullCode, 2022-04-10:
1.5.0 is not ready yet.

Test one of the nighties here : https://github.com/eried/portapack-mayhem/releases

Or take one of the latest bin in test-drive on discord.

The audio problem is fixed in next / nightly

miotislucifugis, 2022-04-10:
Ah ok.   I thought that it was released

closed 2022-04-10 by miotislucifugis

eried, 2022-04-10:
It is released, but it had few small bugs

jLynx, 2022-04-10:
NIghtlys fix this problem. Please report back if you are still having it