portapack-mayhem/mayhem-firmware #563: Screen goes blank when Tx, gets back on at keypress

lu1aax, 2022-04-10:
My screen goes dark every time I Tx. Pressing any key brings it back on. Can't find a way to change this, I don't think this is the intended behavior.

**To Reproduce**
Tx

**Expected behavior**
Screen keeps on when Tx

**Affected versions**
All I tried since 1.4.0

**Additional**
This might be due to my particular portapack version.

GullCode, 2022-04-10:
The update may have activated 'Stealth mode' which is turning of the screen when TX.

Check the icons on top bar, the guy with a hat should be grayed out, in your case it should be green.

Green => stealth , grey => normal

lu1aax, 2022-04-10:
Thanks! That did it. It was not a bug, it was a feature!

GullCode, 2022-04-10:
@ArjanOnwezen  you can close that one ^^

ArjanOnwezen, 2022-04-10:
Thanks @GullCode.. closed

closed 2022-04-10 by ArjanOnwezen