portapack-mayhem/mayhem-firmware #569: Sound settings in audio determine reception

loewal, 2022-04-12:
**Describe the bug**

Release of 12-04-2022.
When trying to receive ADSB, but before accidentally selected SPEC, AM or WFM in audio, there is no reception.
Goes for pocsag too.
Pocsag only works when selecting nfm before in audio, so I believe, that some modes don't have their own settings.
Also the speaker-icon is green, when there is no sound selectable, e.g. adsb.
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'receive'
2. Tap on 'audio'
3. etc...

**Expected behavior**
The right reception.
On ADSB NFM is NOT working.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release
* Previous one (if any) that did not presented the issue
* Old versions available here: https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

jLynx, 2022-04-12:
When you say no reception, do you mean white noise? Also what portapack version are you using? (H1, H2 etc)

jLynx, 2022-04-12:
Can you please add me on Discord so I can send you a custom firmware to test out jLynx#7987

loewal, 2022-04-12:
I am using discord. Do you put it in the testdrive?

jLynx, 2022-04-12:
Send me a DM and I'll send you a custom one

jLynx, 2022-04-12:
Issue is in POCSAG, if you go into audio and select NFM then back into POCSAG it works fine. But if you changed it to SPEC and then went back into POCSAG it does not work

loewal, 2022-04-12:
Correct!

jLynx, 2022-04-12:
The fix will be to just default it back to NFM when you open the app.
Note: Logging should also be off by default

loewal, 2022-05-04:
Hi,
I saw, that you solved the audio-problem.
May I ask again to look at the Pocsag-settings?
They work only, when I set NFM in the audio-app beforehand and remove the ✅  in the Pocsag in "ignore address.."
Can you also default logging to "no"?
Thanks!!

ArjanOnwezen, 2022-05-13:
@loewal  Please check latest nightly tomorrow. Should include fixed modulation. For other (enhancements), please create a separate ticket.

closed 2022-05-13 by ArjanOnwezen