portapack-mayhem/mayhem-firmware #571: Audio off by default, to activate every time

MirkoDalmonte, 2022-04-12:
Installed Mayhem nightly 220412 in H2+ (also 1.4.0=>1.5.0)

Same bug I reported some time ago .. every time I have to click the speaker icon to activate it. After a restart, for example, it is inactive again.
I have to activate the H1 speaker option in Setting => Interface then after each reboot I have to activate the audio using the H1 spesker  icon.

It is not too inconvenient to activate the audio every time but some users may not know how to do it.

GullCode, 2022-04-12:
Do you have a button battery, was it working before, are you sure your button battery is not dead ?

jLynx, 2022-04-12:
@MirkoDalmonte So all settings are loading fine except for the "hide h1 speaker icon" setting?

MirkoDalmonte, 2022-04-12:
Yes, either it is not saved correctly or it is not loaded ...

jLynx, 2022-04-12:
Interesting. I can't replicate this issue myself. Will leave this open for a bit to see if anyone else has this issue

MirkoDalmonte, 2022-04-12:
Sorry I changed my answer because I didn't understand correctly 😬

jLynx, 2022-04-12:
Oh in that case, yes that's correct. That is not a bug. It's just not been added as a feature. So you may want to edit this to be a feature request instead of a bug report

MirkoDalmonte, 2022-04-12:
We also talked about it here:

https://github.com/eried/portapack-mayhem/issues/306#issue-804569628

MirkoDalmonte, 2022-04-12:
> Oh in that case, yes that's correct. That is not a bug. It's just not been added as a feature. So you may want to edit this to be a feature request instead of a bug report

ok, thanks,!!!

xstff78, 2022-04-13:
On my H2 (n_220412), my observed behaviour of the speaker is a bit different:

The “Hide H1 speaker option” is stored and restored correctly after rebooting or start up. The speaker state is not saved but deactivated (grey icon) after rebooting. So far, this is the expected behaviour.

But after rebooting, audio is active despite the grey icon. (Receive -> Audio -> NFM/WFM/AM audible with the corresponding volume). It stays on when activating audio (green icon) and can then be turned off by deactivating audio (grey icon).

From my point of view, the initial behaviour (active despite the grey icon) is a bug.

Can anybody confirm this?

jLynx, 2022-04-13:
@xstff78 the initial behavior you stated is not a bug, we just never implemented storing the on/off value (To my knowledge). But that 2nd part does doesn't like a bug.

But either way this is now a feature request to remember the last speaker state so it will stay on or off depending on what its previous state was.

MirkoDalmonte, 2022-04-13:
> On my H2 (n_220412), my observed behaviour of the speaker is a bit different:
>
> The “Hide H1 speaker option” is stored and restored correctly after rebooting or start up. The speaker state is not [...]

Do you have the H2 or H2+ portapack?
In my H2+ the audio does not remain active, after a restart it is always deactivated, the icon is gray.

xstff78, 2022-04-13:
Just opened the case and checked the version, according to the PCB labelling, it is an H2+. On the back side of the PP, there is a label "PCB v3.4b". I provided a picture here: https://github.com/eried/portapack-mayhem/issues/545#issuecomment-1091150426

 Strange. My speaker icon is also grey after restart, but nevertheless audio is active in this initial state.

MirkoDalmonte, 2022-04-13:
> Just opened the case and checked the version, according to the PCB labelling, it is an H2+. On the back side of the PP, there is a label "PCB v3.4b". I provided a picture here: [#545 (comment)](https://github.com/eried/portapack-mayhem/issues/545#issuecomment-1091150426)
> 
> Strange. My speaker icon is also grey after restart, but nevertheless audio is active in this initial state.

I checked, I observe the behavior you described in the earphones.
The speaker instead is always deactivated.

xstff78, 2022-04-13:
Does this also apply when you unplug the earphones and set the volume to maximum?
I added a poor internal speaker, which produces adequate sound at a volume above ~80.

I agree with jLynx, this issue has now been addressed and should soon be resolved through the feature request.

jLynx, 2022-05-21:
Closed as a duplicate of https://github.com/eried/portapack-mayhem/issues/306

closed 2022-05-21 by jLynx