portapack-mayhem/mayhem-firmware #574: Problem audio portapack H2+ with AK4951

Michel0001, 2022-04-13:
Audio is ok with 1.4.4

Mode : Receive / Audio / WFM

Only noise with 1.5.0 or 1.5.1
Reset auto detection or up at boot not correct that.

Peripherals : RFFC5072 / MAX2837 / Si5351C / AK4951

ArjanOnwezen, 2022-04-13:
@jLynx  I have the same issue on my board (H1), but with same Peripherals.

Michel0001, 2022-04-13:
Not working for my portapack

Slasher006, 2022-04-13:
Does not work for mine eather.

jLynx, 2022-04-13:
Will look into this is the future. I'm taking a step back at the moment

jLynx, 2022-04-13:
@Slasher006 please provide some more information, such as your audio chip and device

mishueen, 2022-04-14:
I have the same problem. WFM noise only.  1.4.4 work ok

arugina, 2022-04-14:
Hi,
I recently bought a PortaPack H1+HackRF One SDR clone, yesterday I flashed under linux the latest Mayhem firmware v.1.5.1 and after restart when I checked out the wide FM band for radio station 102.3Mhz there was just white noise.
Audio chip is the same AK4951 as @Michel0001 
After a dozen of resets and keeping one or the other button down (arrow down, up, left) while reset the problem fixed itself but until the next reset only, then the problem would come back.
I managed **several times** to make it work this way and I suspect some **register initialization "weirdness"** for the **AK4951** chip because I managed to capture this state from the **Debug option** on the PortaPack display .
I attache the registers informationation for the working and for not working state below:
_(I suppose this are the chip registers information, I don't really have much experience with this things)_

`**working**

00 > 40 00 06 40 44 53 0B 14
08 > 00 00 60 00 E1 E1 E1 00
10 > 80 80 80 6E 18 00 00 00
18 > 00 00 0C 01 00 02 B0 1F
20 > 9F 20 00 00 00 00 00 00
28 > 00 00 00 00 00 00 00 00
30 > 00 C1 00 00 00 00 00 00
38 > 00 00 00 00 00 00 00 00
40 > 00 00 00 00 00 00 00 00
48 > 00 00 00 00 00 00 00 00

**not working (this is most of the time)**

00 > 40 00 06 00 44 52 0B 14
08 > 00 00 60 00 E1 E1 E1 00
10 > 80 80 00 6E 18 00 00 00
18 > 00 00 0C 01 00 03 B0 1F
20 > 9F 20 00 00 00 00 00 00
28 > 00 00 00 00 00 00 00 00
30 > 00 C1 00 00 00 00 00 00
38 > 00 00 00 00 00 00 00 00
40 > 00 00 00 00 00 00 00 00
48 > 00 00 00 00 00 00 00 00
`
I am posting in the hope this can help debug and fix the problem. I can post more information if necessary.
Thank you for the great firmware

[image]
[image]
!

ArjanOnwezen, 2022-04-16:
@arugina  please remove battery, including small battery (CR type). This should reset memory

arugina, 2022-04-18:
> @arugina please remove battery, including small battery (CR type). This should reset memory

Thank you for the tip @ArjanOnwezen , I did try removing the battery and several reset's but was random, sometimes woked, other time it did not. Anyhow I flashed one of the "fresh" firmware that I took from discord and the problem is gone.

ytsqkl, 2022-04-23:
> 嗨，我最近买了一个PortaPack H1 + HackRF One SDR克隆，昨天我在Linux下闪烁了最新的Mayhem固件v.1.5.1，当我检查广播电台102.3Mhz的宽FM频段时，重新启动后只有白噪声。音频芯片与@Michel0001的AK4951相同 经过十几次重置并保持一个或另一个按钮向下（向下，向上，向左），同时重置问题自行修复，但直到下一次重置，然后问题就会回来。我设法**多次**使其以这种方式工作，我怀疑**AK4951**芯片的某些**寄存器初始化“奇怪”**，因为我设法从PortaPack显示屏上的**“调试”选项**中捕获了此状态。我在下面附加了工作和未工作状态的寄存器信息：_（我想这是芯片寄存器信息，我对这个东西没有太多经验）_
> 
> `**加工**
> 
> 00 > 40 00 06 40 44 53 0B 14 08 > 00 00 60 00 E1 E1 E1 00 10 > 80 80 80 6E 18 00 00 00 18 > 00 00 0C 01 00 02 B0 1F 20 > 9F 20 00 00 00 00 00 00 28 > 00 00 00 00 00 00 00 00 00 30 > 00 C1 00 00 00 00 00 38 > 00 00 0000 00 00 00 00 40 > 00 00 00 00 00 00 00 00 00 48 > 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 
> **不起作用（这是大多数时间）**
> 
> 00 > 40 00 06 00 44 52 0B 14 08 > 00 00 60 00 E1 E1 E1 00 10 > 80 80 00 6E 18 00 00 00 18 > 00 00 0C 01 00 03 B0 1F 20 > 9F 20 00 00 00 00 00 28 > 00 00 00 00 00 00 00 00 00 00 30 > C1 00 00 00 00 00 00 38 > 00 00 00 0000 00 00 00 00 40 > 00 00 00 00 00 00 00 00 48 > 00 00 00 00 00 00 00 00 00 00 ' 我发布，希望这可以帮助调试和解决问题。如有必要，我可以发布更多信息。感谢您的出色固件！
> 
> 使用手机拍摄的图像。 [image] [image]

> 音频在1.4.4中正常
> 
> 模式 ： 接收 / 音频 / 功率范围
> 
> 只有 1.5.0 或 1.5.1 重置自动检测或启动时的噪音无法纠正这种情况。
> 
> 外围设备 ： RFFC5072 / MAX2837 / Si5351C / AK4951

 I have the same   problem。

jLynx, 2022-05-03:
Created a bug to stop all the duplicates. Please comment on https://github.com/eried/portapack-mayhem/issues/622

closed 2022-05-03 by jLynx