portapack-mayhem/mayhem-firmware #575: Portapack H2 Board

catalinalb, 2022-04-13:
Hello,

Can someone point to the things that we should look for when buying the board? I do have a H2 version from 3 years ago and is not working anymore (display etc...). Should there be a specific chipset or so? Is there a trusted link?

Thank you,

jLynx, 2022-04-13:
Is the display actually broken or does it show a white/black screen on boot?

catalinalb, 2022-04-13:
Hello, it started a few times, and worked normally until I turn it off and then was just white or should I say lit. I even bought a new display from Amazon but no success. I think is some issue with one of the IC's or do you think it can be solved in firmware? Thanks for the interest.

jLynx, 2022-04-13:
This has already been solved. Please read the wiki https://github.com/eried/portapack-mayhem/wiki/Won't-boot

ArjanOnwezen, 2022-04-16:
Please check https://github.com/eried/portapack-mayhem/wiki/Won't-boot

closed 2022-04-16 by ArjanOnwezen