portapack-mayhem/mayhem-firmware #576: issue getting clear audio

Laydenlee, 2022-04-14:
I've got clone portapack H2 and hack RF and I'm running 1.5.0 mayhem. I've got a good looking waterfall and when I use the hack rf function with SDR sharp I can get clear audio but receiving with the h2 is another story I think it's bandwidth but I see other people making it work after flashing firmware
 I've got an ant 500 and the transmit features work fine.

I go receive>audio> 
and I've tried different modulators with different gains VGA and LNA.

eried, 2022-04-15:
@Laydenlee what do you mean by clear audio? can you post a video?

Laydenlee, 2022-04-15:
@eried well no I can't unfortunately but it was just white noise. between now and my first posting the issue I've run mayhem 1.5, 1.5.1 but now that I'm running 1.4.4 it is receiving perfectly. but as I mentioned before I could transmit fine and the RX audio feature shows a good waterfall and I've confirmed my radio works with SDR#. would it be because I have a H2+ ?

eried, 2022-04-16:
Is that 1.4,4 from this github?

Laydenlee, 2022-04-16:
yes I just scrolled down the relece page a ways [here](https://github.com/eried/portapack-mayhem/releases/tag/v1.4.4)

eried, 2022-04-16:
Ok, could you post a video of that old version vs the new with the same source? That will help fixing the problem