portapack-mayhem/mayhem-firmware #577: Problem upgrading the firmware

portasack, 2022-04-14:
Hi all,

I have a Portapack h2+. It worked fine until i tried to update it. I clicked on Flash_portapack_mayhem and the screen went dark. I tried to use all the tricks (dfu etc.) but the screen still wont turn on. Anyone any idea?

jLynx, 2022-04-14:
Please rtfm. Everything is listed in the wiki under troubleshooting

closed 2022-04-14 by jLynx

portasack, 2022-04-14:
I did, many times. All of them didn't work.

eried, 2022-04-15:
Hi @portasack, have you had any luck testing more things?

libratarian, 2022-04-16:
I'm having the same issue. Can't enter dfu mode, reset button does nothing. I took the portapack off the hackrf and hackrf works correctly and enters dfu and can be flashed. Put it all back together and same issue remains, the portapack is essentially bricked. RX and TX lights show faintly and USB shows even more faintly, it will respond to being powered on and off when disconnected from USB but returns to the same state and no device is recognized on both Windows and Linux. This is a severe issue.

eried, 2022-04-16:
What happens if you flash an old version?

portasack, 2022-04-19:
I flashed a Clpd version after hours of search. DFU mode was able to start and flashed it back to 1.4.0. and now it works again