portapack-mayhem/mayhem-firmware #580: Audio not working Portapack H2+ WM8731S

diphtong, 2022-04-14:
Hello,

i have a problem to get audio working.
I have a HackRF ONE with Portapack H2+ from Aliexpress. PCB Version of Portapack is v3.6 (WM8731S) and i installed n_220408.
"Hide H1 Speaker option" is deactivated (X) and the speaker symbol in the status bar is activated (green).

I connected several speakers with no success. Audio output on headphone jack is also not working.
In Debug the WM8731 shows some data.

Thank you in advance!

ArjanOnwezen, 2022-04-14:
Did you try https://github.com/eried/portapack-mayhem/wiki/Won't-boot

diphtong, 2022-04-15:
I did not not write it dont boot...i wrote i get no sound. And yes i tried 
this after firmware upgrade and it works. But no sound

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

WM8731S (Issue #580)
Datum: 15.04.2022, 00:53 Uhr
An: eried/portapack-mayhem ***@***.***
CC: diphtong ***@***.*** ***@***.***> >, Author <

ArjanOnwezen, 2022-04-15:
Did you try pressing and holding  select button during startup? That solved my issue

diphtong, 2022-04-15:
yes i tried...but dont work

GullCode, 2022-04-15:
Can you try my bin in the search app channel in discord? It's just for my own information and to help me hunt that bug.

sivaelid, 2022-04-15:
GullCode FYI.  I did remove the back up battery and started with clean persistent memory to move to 1.5.1. That helped with me  now I seem fine  with your search app and Audio. Simon

diphtong, 2022-04-15:
@GullCode: you mean the calls app? Also no Audio. It seems i have no audio at all.

GullCode, 2022-04-15:
@diphtong no I mean the Search app. The Calls app is badly using 'search' as a keyword for its sources. Anyway it's not really that the app is important, more the mayhem version with it:  it's for me the last working version for both my H1R2 & H2R2. The bin is in the search-app channel, here: https://discord.com/channels/719669764804444213/778568129923383327/964252530018230293

diphtong, 2022-04-16:
[image]

At you link, there is nothing...the site tells me im on a strange location. Or am i too stupid?

eried, 2022-04-16:
I think you have to go thru the discord link in the main page

diphtong, 2022-04-16:
sorry...i dont understand this odd site...

GullCode, 2022-04-16:
Well, here you go then: 
[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/8499938/portapack-h1_h2-mayhem.bin.zip)

GullCode, 2022-04-16:
As @eried stated, you would had to got first through the inscription link for discord to see the topics / channels / links ;-)

diphtong, 2022-04-16:
ahhhh...thank you very much! I will check the next days and give feedback

diphtong, 2022-04-17:
so...i uploaded this firmware and now audio works! Booting with the left button pressed.
I found in options an additional checkbox for touch display....works also. So all fine now! Thank you very much!
Is this the latest firmware, or something older?

ArjanOnwezen, 2022-04-17:
Issues solved

GullCode, 2022-04-17:
@diphtong it's a bit after 1.4.3, at least up to commits on Apr 8, 2022
I'll update to next as soon as next is a working version for all my hardwares, which are H1R2 & H2R2
And it's the freshest bin with THE Search app inside X-D

GullCode, 2022-04-17:
@ArjanOnwezen the issue is solved because my repo's next integration is halted at the last working version for most, or so it looks.
There is still a problem in latest next regarding audio and H2R2.

ArjanOnwezen, 2022-04-17:
@GullCode thanks then I reopen this issue.

diphtong, 2022-04-17:
ah ok, thank you for your information.

sivaelid, 2022-04-17:
HI I have a H2+R2 and this works fine with v1.5.1., and the search ap draft code from Gullcode. The Only issue is that after initial load of firmware you need to hold down the up button. I have had one other case when the PP did not start and had to Hold down Up button at startup. I would not that prior to upgrade to v1.5.1 i did remove the BAtter that suport the persistant memory as this may be corrupted between major versions. - Just my comments

sivaelid, 2022-04-17:
I hope that wen above it was stated as H2R2 it was meant to be H2+R2 as I am not aware of an H2 R2.I one exist can some one let me know what the diference is ?

jLynx, 2022-05-03:
Created a bug to stop all the duplicates. Please comment on https://github.com/eried/portapack-mayhem/issues/622

closed 2022-05-03 by jLynx