portapack-mayhem/mayhem-firmware #581: Add support for certain H2+, maybe related to big QFP100 CPLD support

mgthomas19422, 2022-04-15:
**Is your feature request related to a problem? Please describe.**
The H2+ I recently purchased on ebay just yields a blank screen with this efried firmware. 

I followed the instructions to hold the buttons to detect and change the screen driver, but no luck. I tried both v1.4.3 and v1.5.1 from this repo.

The Mayhem from these sources incorporate a fix, and do work on my unit:
http://www.mmdvm.club/index.php/h2.html  (currently working, running v1.4.3 from here, and as received on the unit)
https://github.com/jumbo77/portapackh2

The description mentions "2021/12 change big QFP100 cpld so need update fw to support it"

Also, at least with this referenced functioning Mayhem v1.4.3  on my hardware, the Mic App is missing.

n3gwg, 2022-04-15:
I just ordered one of these devices and I'd be curious if the proper display driver code can be folded into mayhem too?

mgthomas19422, 2022-04-15:
I believe this issue we are seeing is a duplicate of https://githubhot.com/repo/eried/portapack-mayhem/issues/496

ArjanOnwezen, 2022-04-16:
Please ask for refund, or check with seller
https://github.com/jumbo77/portapackh2

closed 2022-04-16 by ArjanOnwezen

mgthomas19422, 2022-04-16:
Wow, what a rude response! 
It is such a shame all the people getting newer/current H2+ units will not be able to use this wonderful eried portapack-mayhem firmware!

ArjanOnwezen, 2022-04-16:
@mgthomas19422  this firmware is build by volunteers. The more models the more complications. Why not buying a supported model?

sck-nogas, 2022-04-16:
> What a rude response! It is such a shame all the people getting newer/current H2+ units will not be able to use this wonderful eried portapack-mayhem firmware!

Let's look at the facts... Both [HackRF](https://github.com/greatscottgadgets/hackrf/tree/master/hardware) & [Portapack H1](https://github.com/sharebrained/portapack-hackrf/tree/master/hardware) are Open Source Hardware designs, the software for [HackRF](https://github.com/greatscottgadgets/hackrf), [Portapack](https://github.com/sharebrained/portapack-hackrf), and [Portapack-Mayhem](https://github.com/eried/portapack-mayhem) are all Open Source. 

What some people have done is create their own "NOT OPEN SOURCE" version of Portapack Hardware and Software. [They release their own compiled firmware, whenever they feel like it](https://github.com/jumbo77/portapackh2). The volunteer developers here have tried VERY hard to try to figure out the un-documented changes that have been made for the [H2 and H3](https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions) 

So, if you want a working supported Portapack, you need to buy one. OR the Closed Source developer can release their code BACK to the community that they are profiting from.

mgthomas19422, 2022-04-16:
I greatly appreciate the open source contributors working on this, and wish I had the expertise to better facilitate this project.

Between Jumbo77's page and the links below, it sure looks to me like they would like to release a patch back to the community, but are still trying to figure it out, aside from the 4-lines of code change workaround on Jumbo77's page.
http://www.mmdvm.club/index.php/h2.html
https://www.mmdvm.club/usr/uploads/2022/01/2375197635.zip