portapack-mayhem/mayhem-firmware #583: 1.5.1 bug

ytsqkl, 2022-04-16:
**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release
* Previous one (if any) that did not presented the issue
* Old versions available here: https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

ytsqkl, 2022-04-16:
touchscreen faild,audio faild.
v1.4.0  worked well

eried, 2022-04-16:
You need to explain your issue better. Please try again

closed 2022-04-16 by eried

ArjanOnwezen, 2022-04-16:
Please switch touchscreen on in settings - UI settings