portapack-mayhem/mayhem-firmware #584: Audio not working on "classic H1" with FW 1.5.1

UserAZERTY0, 2022-04-16:
A constant blank noise is heard when trying to get audio.
Example: simple civilian WFM radio reception does not give any audio, through headset or speaker.

Nevertheless UI still works fine (waves can be seen and hence stations...)
When recording audio via wav files, wav are fine through playback (audio well recorded).

As such, it seems it comes from how the audio is addressed with the new firmware (did not test with 1.5.0 but until 1.4.4 I think audio was ok) - maybe in correlation with how it was tried to managed new audio component (H2...)

ArjanOnwezen, 2022-04-16:
Please check https://github.com/eried/portapack-mayhem/wiki/Won't-boot

ArjanOnwezen, 2022-04-16:
Hold select while booting - check audio
Hold up whole booting - check audio
.... Etc

UserAZERTY0, 2022-04-16:
Thanks, it worked: pressing select/ok (central button) at startup did the trick.
If kept pressed during the whole boot, it does not work though.

Still I find strange that with the classical/historical H1 I have to do tricks, as in my pov "default boot" should at least cover "default hardware".

ArjanOnwezen, 2022-04-21:
Issue solved.

closed 2022-04-21 by ArjanOnwezen

UserAZERTY0, 2022-04-22:
Well every time I want to use my Portapack I have to reboot and test it with civilian radio several times (and what if I cannot receive civilian radio? I would then have to test it with a known/generated radio source)
Frankly I still consider as a bug that the default boot cannot manage anymore the default hardware ☹

jLynx, 2022-05-03:
Created a bug to stop all the duplicates. Please comment on https://github.com/eried/portapack-mayhem/issues/622