portapack-mayhem/mayhem-firmware #586: Problem a clone of HackRF + Portapack H2 the buttons do not light up, the PC does not detect the device

batubek, 2022-04-17:
Greetings
I bought a clone of HackRF + Portapack H2, flashed with mayhem firmware version 1.4.3, everything worked properly. One day, when reinstalling the drivers on the PC, HackRF + Portapack H2 turned off and no longer turns on, the buttons do not light up, the PC does not detect the device in any way. DFU mode is not activated. However, if you disconnect the boards and connect only HackRF, everything works. I downloaded the latest firmware and flashed HackRF first in DFU mode, then flash_hackrf_one.bat. HackRF + Portapack H2 does not work at all, followed the instructions, black screen from wiki, result 0.
If you flash HackRF without Portapack H2 with flash_portapack_mayhem.bat firmware, HackRF stops working and you need to restore it via DFU mode and flash_hackrf_one.bat firmware. Please help anyone who faced this problem and solved it.

jumbo5566, 2022-04-20:
HI,pls ensure your Portack H2+ is TQFP64pin  or TQFP100 pin?

batubek, 2022-04-20:
Hello, I am writing through a translator. The reverse translation is incorrect. I can't understand what you mean (TQFP64pin TQFP 100 pin). I posted photos of my HackRF Portapack H2 boards.
[https://drive.google.com/open?id=1Dj8DKVUsRVL0JmU7rvriuxTVidFqYFZ7&authuser=batubek%40gmail.com&usp=drive_fs](url)

ArjanOnwezen, 2022-04-25:
@batubek  please check https://github.com/eried/portapack-mayhem/wiki/Won't-boot

batubek, 2022-04-25:
@ArjanOnwezen 

> please check https://github.com/eried/portapack-mayhem/wiki/Won't-boot

> followed the instructions, black screen from wiki, result 0

ArjanOnwezen, 2022-04-25:
Did you press up while rebooting
Press down while rebooting
Press left while rebooting
Press middle button while rebooting
Did all these options fail

ArjanOnwezen, 2022-05-01:
@batubek did this work?

batubek, 2022-05-04:
> did this work?

No

batubek, 2022-05-04:
@ArjanOnwezen 
Why did you close the discussion of my problem, it was not resolved, I wrote to you that the result is 0, but you closed the topic.

batubek, 2022-05-04:
@jLynx 
Thanks

jLynx, 2022-05-04:
So the portapack does not work even when plugged into a battery bank instead of the computer?

ArjanOnwezen, 2022-05-04:
> @ArjanOnwezen Why did you close the discussion of my problem, it was not resolved, I wrote to you that the result is 0, but you closed the topic.

Oops.. sorry for that. Wrong button.

batubek, 2022-05-05:
@jLynx 

> So the portapack does not work even when plugged into a battery bank instead of the computer?

1. You are right, portapack does not work with the battery connected. Indicator LEDs do not turn on and does not respond to key presses.
2. If you disconnect the battery and assemble the portapack, connect it to USB, it also does not work, similar behavior with or without a battery.
3. I wanted to try charging the battery with a third-party charger and see if it works. Maybe the battery charging circuit is burned out.
P.S. HackRF itself, in a state disconnected from portapack, is fully operational.

arturmaciag, 2022-05-11:
I think that your portapack needs modified firmware because the Hackrf clone used another CPLC (IC TQFP 100 or 64).
You have to modify firmware from this repository or use binary from the seller (version 1.4.3).

ArjanOnwezen, 2022-05-11:
@batubek  Have you tried latest nightly. It looks like it's pretty device specific.

tony1492, 2022-06-02:
I was able to get mine working with a long press of the the down button before turning the portapack on,  so hold down the down button then turn the portapack on aknd keep holding the button till it comes on mine took like ten seconds. 
Tony

jLynx, 2022-06-02:
@batubek are you still having this issue? If so are you on our discord server? I can send you a custom firmware to test that should solve your issue

batubek, 2022-06-03:
@jLynx 
> @batubek are you still having this issue? If so are you on our discord server? I can send you a custom firmware to test that should solve your issue
The issue is still there!!! No, I am not a member of your discord channel. You can send me the firmware or a link to it by e-mail. I will be happy to test this firmware. Thank you for helping.

CrazyM0nkey, 2022-06-08:
@batubek I had similar problems after the update. First it didn't turn off, than it didn't boot. The fix in my case was pretty easy. 
I had to dissconnect the battery from the board. Since than it is working fine with FW 1.5.3.

jLynx, 2022-06-08:
@batubek sorry I forgot about this. Can you please try the latest nightly release. That should include the fix

batubek, 2022-06-10:
@jLynx 
> sorry I forgot about this. Can you please try the latest nightly release. That should include the fix

Does not work
maybe I'm not following the sequence of steps on the firmware correctly?
Please note that in my connected version HackRF + Portapack H2 does not work and the computer does not see the connection in the devices, it is not displayed in any way. Therefore, I first flash the HackRF board separately with the flash_hackrf_one.bat file, then update the contents of the Micro SD, then attach the Portapack H2 to the HackRF board and try to start from the button. I also tried flashing HackRF separately with the flash_portapack_mayhem.bat file and doing the above procedure, but the result is the same and plus everything HackRF stops working as a separate board has to be restored through DFU mode.

jLynx, 2022-06-10:
What you will need to do is run the dfu......bat file and then once that is done you should then be able to run the mayhem.......bat file to flash it. If your computer is still not detecting it, then make sure you have the drivers installed

batubek, 2022-06-10:
@jLynx 

> What you will need to do is run the dfu......bat file and then once that is done you should then be able to run the mayhem.......bat file to flash it. If your computer is still not detecting it, then make sure you have the drivers installed

I have drivers installed.
The HackRF board without the connected Portapack H2 module works, the computer detects it, displays connected devices in the section, the SDRSharp program works correctly. However, if you connect the Portapack H2 board to HackRF, the computer stops seeing the device, when connected, there is no reaction, it is not displayed in the device manager in any way, including as an unknown or hidden device. Therefore, I had a question for you how to flash the device step by step.

Brumi-2021, 2022-06-11:
Hi @batubek ,  I want to believe that you have big CPLD TQFP100 in the Portapack with battery under H2 or H2+ and it is also giving me a lot of issues... 
(So far I unbridled mine (H2+ big CPLD )  many times  , but always dissasembling Hackrf brd from Portapack, and this is not friendly at all . I am trying to find the way to avoid that nasty inconveniences . Maybe @jumbo5566  knows other better ways ). The process is easy but you need single separated Hackrf. 

Anyway , can you share here a photo of the top view of the Portapack board .  (We need to know the CPLD type and if possible the printed  name of that Portapack brd, example H2+ version 3.6 )  
Other question , are you using linux ?

Brumi-2021, 2022-06-11:
> I think that your portapack needs modified firmware because the Hackrf clone used another CPLC (IC TQFP 100 or 64). You have to modify firmware from this repository or use binary from the seller (modified version 1.4.3).

Hi @arturmaciag , did you manage to modify your fw repo ?  I also did about the first black screen after reset ,  in the Portapack app part , (that now is integrated in the 1.5.x , )   this is ok and no problem . But when I try to integrate locally the second change (about black screen in Hackrf usb mode )  in the submodule Hackrf fw , I am having problems ... only once worked , now recently not . and I would appreciate help.  Are also in discord ?

batubek, 2022-06-21:
@Brumi-2021

> Hi @batubek , I want to believe that you have big CPLD TQFP100 in the Portapack with battery under H2 or H2+ and it is also giving me a lot of issues... (So far I unbricked mine (H2+ big CPLD QFP100 ) many times , but always dissasembling Hackrf brd from Portapack, and this is not friendly at all . I am trying to find the way to avoid that nasty inconveniences . Maybe @jumbo5566 knows other better ways ). The process is easy but you need single separated Hackrf.
> 
> Anyway , can you share here with us , here, a photo of the top view of the Portapack board . (We need to know the CPLD type ,QFP100/QFP64 and if possible the printed name of that Portapack brd, example H2+ version 3.6 ) Other question , are you using linux ?
> 
> The process that I normally use to recover mine is the last part of #651. I also believe that you can skip the step (6) of updating Hackrf version (this is just very rare case ) This I did only once because mine initially had a very old Hackrf CPLD fw version from the vendor ) . And additionally after reflashed (using dfu-util and hackrf_spiflash, be aware that from Mayhem 1 5.3. onwards (that supports both Portapack-CPLD's) . In the first boot , you need to follow the process described , in the wiki "won't boot" ( https://github.com/eried/portapack-mayhem/wiki/Won't-boot) pressing left key (in my case ) while power on , and wait 10 secs, but only once (but you need to have Portapack with cell battery -to support persistent ram- memory or the usual micro SD on the Hackrf side, to store the Portapack type that you have in next boots )

https://drive.google.com/drive/folders/1Dj8DKVUsRVL0JmU7rvriuxTVidFqYFZ7?usp=sharing

Brumi-2021, 2022-06-22:
Hi ,looking your photos , it seems you have a H2 with standard TQFP64 CPLD Portapack . So, basically I understand that you do not need any special fw from vendor . The fw 1.4.3 that you installed I guess it was  the official Mayhem 1 4 3 . Then if my assumptions are correct, if recently you can not see the device from the pc when you are in hackrfmode ,->make sure that it is not your driver PC problem (best case , checking with other or linux system) or other possibility is that the written fw on flash had some corrupted part in that hackrfmode area  (in that case you need to try to reflash from the Hackrf board alone, through DFU and reflashing some Mayhem official version)

batubek, 2022-06-22:
@Brumi-2021
Hello, thanks for your help.
Mayhem 1 4 3 firmware was from this source from the release section, the last one at that time, after installing this firmware, my HackRF + Portapack H2 worked fine. Connected to the PC was displayed in the device manager. The problems started when I decided to reinstall the driver to run SDRSharp, namely when installing zadig.exe. The device randomly turned off and won't turn back on. When the boards are open, the HackRF board works with PC and phone no problem. When HackRF + Portapack H2 boards are connected nothing happens everything looks like a brick.
I am translating what you write with the help of a translator, given that we speak a technical language, the translation may not be correct, so I ask you to explain in short and as simple as possible points how to flash HackRF separately without a Portapack H2 board connected, with Mayhem firmware. All my attempts lead to the inoperability of HackRF and I have to restore the firmware via DFU.
1.
2.
3.
Explain step by step.

Brumi-2021, 2023-05-21:
Hello @batubek , did you manage to solve it,?
Can we close that problem,? 
Of you need extra help ? You can also check the latest lrelated wiki post ,,

KimIV, 2023-05-24:
Hello everyone I ran into a similar problem after updating the firmware to version 1.7.1 and trying to check the availability of microSD from a computer via USB. I turned on this mode in the utilities on the device and waited a little. After about a minute, the device turned itself off and did not turn on again.
The following facts were revealed:
- Separately, HackRF One works fine. Seen by a computer running Windows 7.
- In connection with the PortaPack H2, the device does not turn on by any combination of buttons and is not visible by the
 computer.
- The voltage at the battery connector when the battery is connected is slightly more than 2 volts.
- The voltage on the battery, if it is turned off, is 3.7 volts.
- If you connect a laboratory power supply instead of a battery, the current consumption will be slightly more than 1 A.
- At the same time, the ETA6102 chip and a large resistor standing next to it are noticeably heated.
[image]
[image]

QGEEE, 2023-06-24:
@KimIV were you able to resolve this issue? I am having similar issues with fw 1.7.1 and 1.7.2

KimIV, 2023-11-24:
> @KimIV смогли ли вы решить эту проблему? У меня похожие проблемы с fw 1.7.1 и 1.7.2

Yes, I ordered on AliExpress ETA6102. When he arrived, he soldered it with a hair dryer. Now my device is working fine!

6Shmoney9, 2024-02-11:
My screen has bin blue boot mode sent I bought it it ses it's working fine but right now it's a expensive door stopper help I feel like I did everything inm stopped but not that stopped help