portapack-mayhem/mayhem-firmware #588: H1 white noise on 22-04-18

loewal, 2022-04-18:
**Describe the bug**
White noise

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Audio'
2. select WFM
3. white noise

**Expected behavior**
Broadcast FM reception

**Affected versions**
22-04-18

22-04-12 Worked!!

naquaii, 2022-04-20:
Similar issue. I only get white noise but I can hear there's something there when tuning to a WFM radio station.
Works fine in 1.4.4 stable but in 1.5.0 stable and later it does not.

Other receivers other than audio seem to work fine.
I have an original H1 portapack.

UserAZERTY0, 2022-04-22:
Probably same bug as the one I mentioned here: https://github.com/eried/portapack-mayhem/issues/584#issue-1206190649

jLynx, 2022-05-03:
Created a bug to stop all the duplicates. Please comment on https://github.com/eried/portapack-mayhem/issues/622

closed 2022-05-03 by jLynx