portapack-mayhem/mayhem-firmware #594: Mayhem v1.5.1 - Portapack display not working properly

NullVibes, 2022-04-20:
**Describe the bug**
After re-flashing portapacks with mayhem firmware v1.5.1, the portapack display frequently does not work upon applying power.  Tested on 6 different portapacks, (with 6 different HackRFs), all brand-new, in the last 24 hours.  I've re-flashed with & without using DFU-mode to clear the present firmware.  x5 units are able to restore the display after pressing the Reset button 3-5 times.  x1 unit does not restore display until resetting firmware in DFU-mode.

**To Reproduce**
Steps to reproduce the behavior:
1.  Apply power while pressing the inboard button to enter DFU-mode.
2. At the terminal:  dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
3. Without powering-down, then issue:  hackrf_spiflash -w portapack-h1_h2-mayhem.bin
4. Reset power

**Expected behavior**
Upon applying power, either the Mayhem splash-screen or button list should be displayed.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* v1.5.1 affected
* v1.4.7(?) on one device did not have this issue.

zxkmm, 2022-04-21:
https://github.com/eried/portapack-mayhem/wiki/Won't-boot

ArjanOnwezen, 2022-04-21:
Please try page above..
Hold select while booting - check audio
Hold up whole booting - check audio
.... Etc

@NullVibes if it's not working.. ticket will be reopened

closed 2022-04-21 by ArjanOnwezen