portapack-mayhem/mayhem-firmware #604: HackRF One Portapack H1 bricked after 1.5.1 update

raguini, 2022-04-22:
**Describe the bug**
Installed firmware 1.5.1.  After update, screen flashes white then goes black.  HackRF One with Portapack no longer recognized by Windows 10.   DFU update gives the following:

*** Run HackRF firmware in RAM via LPC DFU ***

This is used to "unbrick" your HackRF, if you are no longer able to use
HackRF tools to flash or operate your HackRF.

Connect your HackRF One to a USB port on your computer.

Hold down both the DFU and RESET buttons on the HackRF.
Then release the RESET button (closest to the edge).
Then release the DFU button.

Press any key to continue . . .

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Cannot open DFU device 1fc9:000c
No DFU capable USB device available

Press any key to continue . . .

Following recovery procedures for 'white screen' and 'black screen' issues has not affect.

olemlong, 2022-04-24:
had the same issue. This solved the issue for me =) https://github.com/eried/portapack-mayhem/wiki/Won't-boot

ArjanOnwezen, 2022-04-25:
@raguini  please try page mentioned above.. Closing the issue. Please comment if you still experience issues.

closed 2022-04-25 by ArjanOnwezen