portapack-mayhem/mayhem-firmware #605: Soundbard app killed the amp by constant TX with Amp enabled?

zviratko, 2022-04-24:
Today I wanted to demonstrate a simple FM transmission using the Soundboard app. At first it worked in my testing but shortly after it failed to TX anything.
After I got home I investigated, and it seems that as soon as the Amp is enabled in the Soundboard settings (without presssing START!), it actually tries to TX? At least the TX LED turns on, though it doesn't seem to TX on the set frequency. Is that just an indicator turning on errorneously, or is it TXing something, somewhere? 
In any case, my Amp is dead now (lower TX power with it enabled), which might or might not be related to this. But assuming the TX LED indication is correct then it could have killed it quite easily as I was swapping antennas and stuff while in the app, possibly (likely) with amp setting enabled...

zviratko, 2022-04-26:
Not directly, I just know it doesn't transmit on the indicated frequency, but it's possible it transmits elsewhere. Nevertheless, if it does turn on the TX amp then with it dead I likely won't be able to confirm anything anymore.
But let me know what I can try - I can measure it with a simple multimeter or something like that.

ArjanOnwezen, 2022-04-27:
You could indeed see if there's output when playing a file using soundboard. Then try the same with amp on before playing any file.

ArjanOnwezen, 2022-05-13:
I can't reproduce this on latest (nightly) version.
Here it only transmits when pressing START

closed 2022-05-13 by ArjanOnwezen