portapack-mayhem/mayhem-firmware #607: Battery statistics window

n3gwg, 2022-04-25:
Problem: No reasonable manner by which to determine the residual battery capacity left as the LED rotating dial method is highly inaccurate.

Solution: Add a screen which depicts the current battery voltage, percentage of battery charge residual to the battery pack, and battery temperature in Celsius and Fahrenheit. Given the crowd that uses this device, perhaps it would nice to display it in Kelvin as well! :)

eried, 2022-04-25:
It is not possible with the current hardware, sorry :/

closed 2022-04-25 by eried

n3gwg, 2022-04-25:
Well, it is rather difficult to argue with hardware limitations, though I hope future revisions of the board design become inclusive of such functionality.

Thank you for your most brisk response.