portapack-mayhem/mayhem-firmware #608: automatic setting of time and date via radio signal

n3gwg, 2022-04-25:
Problem: setting the time and date via a known time source

Solution: Add functionality to the firmware such that it could learn the time and date and set it as such via either GPS signal reception or via WWV

ArjanOnwezen, 2022-04-26:
Not possible with current hardware

eried, 2022-04-26:
(what it could be possible is to read those radio stations https://en.wikipedia.org/wiki/Radio_clock maybe?)

ArjanOnwezen, 2022-04-26:
That frequency is too low for the receiver.

GullCode, 2022-04-26:
While @ArjanOnwezen is true for a few countries, it should still be usable for a lot.
http://www.bonne-mesure.com/horloge_radio-pilotee.php
It's in French but you can see some are emitting over 1Mhz

ArjanOnwezen, 2022-04-26:
Ones above 1MHz are very local examples.

n3gwg, 2022-04-26:
Arjan,

I am a bit confused. You seem to suggest that current hardware is incapable of receiving a GPS signal? If the hardware can receive a GPS signal, why could the proper time not be acquired from that signal? Every GPS I have ever owned has been able to do that. I am sure I am misunderstanding something, but can you kindly tell me what I am not realizing?

Stuart

sivaelid, 2022-04-26:
n3gwg  While the  L1 1575MHz and the L2 1228MH  frequencis can be received you need processing power to decode the signal,   key the position data etc. The PP does not have that capacity for this  and it does not have a dedecated Decode chip for GPRS.The HackRF while over 6 years old in design is still a good SDR if supported with good PC.

n3gwg, 2022-04-26:
I am a bit confused, what does GPRS have to do with GPS? Lest I am misunderstanding something, GPS is Global Positioning System, GPRS is Ground Penetrating Radar System. I was not asking about GPRS. What are you trying to suggest conflating the two? I see no relation myself betwixt GPRS and GPS. Can you disambiguate?

My next question regarding the setting of the clock, is there a manner by which the device might be able to obtain its clock and date setting via NTP when plugged into a computer via USB (say after a firmware update)?

Stuart

sivaelid, 2022-04-26:
n3gwg My error typing too fast. IT was GPS chip not GPRS. And on your second point yes if NTP On computer has Time signal then Yes potentially you could Update the time frrm the computer , Are you going to write the code??. I do think you need to understand thAT PP has limited resources and the code in run on the cores of the HackRF. If a lot bigger CPU /Mem etc then it could do a lot more complex decoding.. E.g use the HackRF connected to the computer  and RUn SDR/GNU programs.

eried, 2022-04-26:
Maybe decoding pieces of the GPS signals to get the datetime is not actually a bad idea, it could be even easier than those old school radio stations 🤔

Brumi-2021, 2022-04-26:
Other option might be to get the clock from RDS (subcarrier 57khz)  data inside the FM+ RDS .
But not sure , if it is a common World wide FM protocol standard or just EU  ? 
https://en.m.wikipedia.org/wiki/Radio_Data_System

n3gwg, 2022-04-26:
I do know that FM + RDS is in the US, but I have no idea if it is inclusive of any date/time information. I'd imagine that FM car stereos are reasonable standard worldwide though.

voxo22, 2022-04-26:
Yes, some FM stations trasmit clock info within RDS, but not all. While not using portapack, I was able sucessfully decode any time signal (DCF77, MSF, BBC, FLE) with hackRF connected to PC, proper software and, of course, proper antenna (long wire in this case)...

Dirk275, 2022-04-28:
By the way:
There is a place for a GPS "chip" on the PortaPack (at least H1):
https://github.com/eried/portapack-mayhem/issues/259
Adding this chip could give date, time, location infos ...

ArjanOnwezen, 2022-04-28:
> By the way: There is a place for a GPS "chip" on the PortaPack (at least H1): #259 Adding this chip could give date, time, location infos ...

Sure but that is not a common solution.
Sone people already did this, but source is unavailable

n3gwg, 2022-04-28:
It is worthy of notation that indeed I did purchase a PortaPack H2 (granted at the time I ordered I had no idea of which PortaPacks are developer sanctioned or not and am still not fully clear on that).

The foregoing notwithstanding, I do hope that all versions of the PortaPack in the future are inclusive of a GPS chip as the additional cost is likely rather modest and the functional gains are surely interesting (time, date, location). In fact, it might be possible to add a function to the PortaPack as well (if it had GPS) to be able to track how strong a signal is over time when moving along and also recording GPS coordinates. So if one is driving 25 mph it could be observed that RSSI of x frequency increased not just when you drove west, but, when your elevation began to increase over z feet above sea level as you traveled at x speed.

keagan73, 2022-04-28:
maybe you can use the FM RDS i have a eton short wave radio that can set its time that way.

janicektrnecka, 2022-04-30:
FM and RDS should be easy enough to be processed by internal chipset... (although I have absolutely no idea how to do it).

gullradriel, 2023-05-20:
@ArjanOnwezen was surely right since no one came up with a good solution.
Anyway it does not look like a useful feature and would need special ham it up things since signals below 10MHz are shit on the hackrf.
Closing until someone come with a PR and a proof of concept.

closed 2023-05-20 by gullradriel

yath, 2025-01-13:
If you just want to set the time and don’t necessarily want to do it via RF:

```
stty -F /dev/ttyACM0 115200 -echo -icanon onlcr
cat /dev/ttyACM0 > /dev/null &
date +"rtcset %Y %m %d %H %M %S" > /dev/ttyACM0
```

gullradriel, 2025-01-13:
> If you just want to set the time and don’t necessarily want to do it via RF:
> 
> ```
> stty -F /dev/ttyACM0 115200 -echo -icanon onlcr
> cat /dev/ttyACM0 > /dev/null &
> date +"rtcset %Y %m %d %H %M %S" > /dev/ttyACM0
> ```

Precision for that answer: it's only in case you are using USB Serial with your unit. Description of the possible commands is available in the [wiki/USB-Serial-Console](https://github.com/portapack-mayhem/mayhem-firmware/wiki/USB-Serial-Console) . USB Serial can be used in a terminal app, and also on the [hackrf.app ](https://hackrf.app/) website.

Precision for [hackrf.app ](https://hackrf.app/)  : it's automatically putting the device at the current date and time on device connection.