portapack-mayhem/mayhem-firmware #614: bandwidth filter

keagan73, 2022-04-28:
it would be reallly nice to be able to size the bandwidth filter in the AM mode

GullCode, 2022-04-29:
What do you mean ? If you go on bw and use the rotary encoder you should be able to switch between LSB USB DSB (maybe others I don't remember)

Brumi-2021, 2022-04-29:
Maybe he wants to say , to be able to adjust BW filter  of the AM (i.e  3khz, 5khz ...) . Usually in other PC SDR programs like sdrangel, SDR#, GQRX ... I think is possible just moving the mouse cursor .

Brumi-2021, 2022-11-22:
Hi @keagan73 ,
Recently we just added two different AM bandwidth filter selection (DSB 9 / DSB 6k ) , means 4,5 Khz  and 3khz demodulate AM  audio baseband respectively.  (PR #747 )

can we close that issue ?

keagan73, 2022-11-23:
Yeah got right ahead. Is it in the new nightly?

Brumi-2021, 2022-11-23:
Hi yes , @keagan73 
you can test it from Nightly release - 2022-11-21 onwards .  Pls check it and comment here , if we can close that issue . (It is applied on Audio RX / Reception part of MIC  / scanner / Recon App ) .

Brumi-2021, 2022-11-30:
Hi @keagan73 , have you had any chance to test that new  AM switchable bandwidth filter feature ? 

Can we close that issue ?

Now we  have two AM switchable filters,  But i think enough  :   (DSB 9k / DSB 6k ) , means 4,5 Khz and 3khz demodulated AM Baseband . 

If needed in future , it can be easily add some more , like (DSB 11k  , means 5,5khz audio  baseband ,  but I think not necessary ,  because  not so much acoustic difference to DSB 9k.

(As the audio channel sampling rate is running at 12 Khz , the current easy limit is <6khz )

we will wait about your feedback a few days more , but 
if no other objections, we will close that issue soon.
  Thanks

keagan73, 2022-12-02:
Yes I have and it has been working great. Go right ahead and close it.
Thank you for helping improve it, it's very much appreciated if there's
anything I can do to help let me know.

Brumi-2021, 2022-12-02:
Hi @keagan73 , thanks a lot for your test & validation .
Glad to see that you also like it ! 👍
Let’s close that issue and enjoy about it !  
And continue working /learning with more new  features .
73 , cheers

closed 2022-12-02 by Brumi-2021