portapack-mayhem/mayhem-firmware #615: weather balloons

keagan73, 2022-04-28:
it would be nice to be able to decode the LMS6 radiosonde with the mark 2A telemetry. i live in the usa and that is the one we use. i can provide super clean audio of it. i have both 403mhz and the 1.7g radiosondes. i will use sonde.exe and sondeMonitor when tracking them down.

joyel24, 2023-05-02:
> it would be nice to be able to decode the LMS6 radiosonde with the mark 2A telemetry. i live in the usa and that is the one we use. i can provide super clean audio of it. i have both 403mhz and the 1.7g radiosondes. i will use sonde.exe and sondeMonitor when tracking them down.

If you are able to ship to France at least one LMS6 and buy me an RTL-SDR I'll be able to begin to work on it.

DiX880, 2023-10-23:
Is it impossible to make support for the probe (MeteoRadiy MRZ-N1 (400 MHz)) ?