portapack-mayhem/mayhem-firmware #616: Problem upgrading the firmware

hackrf-user2017, 2022-04-30:
up to version 1.4.4 everything is well stitched and working, starting from version 1.5.0-1.5.1 after the firmware there is no image and nothing works. My version of PortaPack H2

janicektrnecka, 2022-04-30:
Have you tried loading different LCD driver ? ((holding specific buttons during startup).

tita2000, 2022-04-30:
same here with a portapack h2+ and a display labeled Z320IT008 V0.0 2020-07-13. The software works as I can enter HackRF mode by pressing 5x down 1x right 1x center 1x center. I also tried pressing up or doen while booting, but that did not work.

Brumi-2021, 2022-05-01:
Hi , did you follow the release guide ?
https://github.com/eried/portapack-mayhem/wiki/Won't-boot

tita2000, 2022-05-01:
> I also tried pressing up or down while booting, but that did not work.

I keept pressing up while switching on and tried the even the down. Also tried the ok for reset. For the left I do not exactly know which prodedure gets invoked and how it works.

Brumi-2021, 2022-05-01:
I do not have yet , any H2+ (soon I will have one similar as yours) . But I can see that most of new H2+ Portapack board sellers  are now using a new CPLD QFP100 , instead of the original QFP64 . Then it seems that the fw needs to apply different boot code in one case or the other. And I think that is why left key while booting is used . (This is my current understanding) .

hackrf-user2017, 2022-05-01:
> Hi , did you follow the release guide ? https://github.com/eried/portapack-mayhem/wiki/Won't-boot

Hello, thanks for this tip, it worked for me, holding the up button when turning it on. Thank you.

closed 2022-05-01 by hackrf-user2017