portapack-mayhem/mayhem-firmware #620: No audio in Mic TX app

diphtong, 2022-05-02:
Hello,
i found some strange behaviour on audio.
Hearing audio with Audio app works. When i switch then to Mic TX app and activate "RX audio listening", there is no audio (on same freq). Switching back to Audio app shows me the spectrum, but no audio (but audio is still activated).
It seems the Mic TX app will deactivate audio in some way.

**Affected versions**
- HW: Portapack H2+ PCB v3.6 (WM3871S)
- tested firmwares: - https://github.com/eried/portapack-mayhem/issues/580
                               - v1.51
                           
Thank you!

Brumi-2021, 2022-05-02:
Hi @diphtong , 
There is also another user with similar problem, and might be related to the ALC introduction on AK4951 platform (that should be transparent to the WM8731 platform)
which CPLD do you have in your Portapack ? The original QFP64 or the new version QFP100, bigger size ?

TheDaylightWalker, 2022-05-02:
I also have this issue with any version above 1.4.3 with WM audio chip H2

deano817, 2022-05-02:
I concur no audio in tx above version 1.4.3 with my h1.

Brumi-2021, 2022-05-02:
Hi @deano817 , which platform do you have? I guess H1 WM8731 , not ? 
Which Portapack CPLD  version QFP100 or QPF64 ? (maybe not related , but better to point it ) .
In the rest of  receiver App you got correct audio,  in the headphones ? 
And when you go to Mic App , and you select "RX audio listening"  , in Mic App , no headphone audio ? 
And no Mic vumeter indication when talking close to the Mic in reception or transmission?

Brumi-2021, 2022-05-02:
Maybe I misunderstood you .
And you have correct audio in the "Rx audio listening" inside Mic App , but somehow ,  it deactivate the audio of the rest of Receiver App after using Mic App  ?

deano817, 2022-05-02:
I haven't looked at to see which platform or clpd I have as I have 5 of them and they all behaved the same until I went back to 1.4.3 then everything was working again except for ctcss and roger beep. I've been in using the tx app for tx for now and haven't tested rx.

diphtong, 2022-05-03:
@Brumi-2021: 
1: no VU-meter amplitude before pressing the TX button when talking to the mic (completely grey VU-meter)
2: i have the QFP100

diphtong, 2022-05-03:
@Brumi-2021: correct, when i select "RX audio listening", no audio. And also no VU-meter indication.
When i then go out of the Mic app and then go to the Audio app, no sound. But before going to the Mic app, audio worked in the Audio app.
Rebooting the Portapack audio in Audio app works again.
It seems the Mic app kills the audio completely in background, because the speaker symbol is always green.

diphtong, 2022-05-03:
steps to reproduce:
- power on portapack
- activate speaker
- go to Receivers -> Audio
- select a frequency of a radio station, audio is OK
- go back to main menu
- go to Mic app -> no audio, no VU-meter indication
- go back to main menu
- go to Receivers
- go to Audio app -> no Audio output anymore

Brumi-2021, 2022-05-03:
Ok @diphtong thanks for your confirmation.
I already order last week a second one Hackrf from china , this time, with bigger screen (H2 + WM8731 + QPF100), to be able to pre-test at home any new PR release , and eventually solve that problem . 
Apologize for the inconveniences. 
If you are in rush , in next days I can try to release a binary 1.5.1. , just removing temporary the  ALC PR feature to confirm that it is the root cause. (that theoretically it should be transparent to the WM8731 platform ) . Or let me wait till getting and debugging when I got the new sample . (B/june is telling the web) 
Do not worry , we will investigate and fix it !

diphtong, 2022-05-04:
Ok, thank you very much!
Im not in rush, so lets wait for your new hackrf and take your time to fix it.
But just or my information...what is the "ALC PR" feature and what is it doing?

Brumi-2021, 2022-05-04:
Hi @diphtong 
I added ALC feature based on opened issue #465 from 1.4.4. Onwards. 
And recently I also submitted another commit reactivating the lost CTCSS , and roger beep ... (still not merged) PR #621  
This is the recent summary feature table of this Mic App .
If you have correct sound till 1.4 3 , big probabilities that there is some side effect about ALC feature in your platform.  Most likely this weekend I will try to create special test version for you , same as #621 but removing temporary the ALC feature to confirm the origin of your problem . Thanks 
[image]

diphtong, 2022-05-04:
Ok thank you very much for explanation and support!

Brumi-2021, 2022-05-04:
Hi @diphtong ,   
I could generate the testing binary much easy than i expected. 
Pls. find attached , that binary  test  ,  that I have build from my latest PR #621  (based on 1.5.1 )   (but I removed the previous added  Auto mic Level Control feature for the AK4951, and theoretically  transparent to the WM8731 platform  in the PR #497, to confirm or discard the side effect in your device ) .

Pls. unzip the file , and test it .  (I change the name :  portapack-h1_h2-mayhem_1_5_1_no_ALC.bin)

And let us know if in your platform  H2 + WM8731 + Portapack CPLD_QFP100 , 
if you have correct  sound in Mic App as  fw 1.4.3  , with correct  vu-meter , correct   FM / AM / USB /LSB/DSB audio modulation  + correct  CTCSS / ROGER BEEP / Correct MIC GAIN  and correct audio in other receiver app after using Mic app. 

If all  works well in your platform , as good as the previous fw 1.4.3,   we should   merge that version (removing ALC temporary  ) till  debugging with both platforms  and fix all your  side effects.   
[portapack-h1_h2-mayhem_1_5_1_no_ALC.zip](https://github.com/eried/portapack-mayhem/files/8627066/portapack-h1_h2-mayhem_1_5_1_no_ALC.zip)

Waiting for your news .

diphtong, 2022-05-05:
Good morning!
Thank you very much!
I uploaded the firmware and can give feedback:

    correct vu-meter ,(just talking close to the MIC even no transmitting ) -> yes, its working now (dont need to be close to the mic, very sensitive)
    correct FM / AM / USB /LSB/DSB audio modulation when TX .-> works
    correct CTCSS / ROGER BEEP / Correct MIC GAIN when TX -> works
    Correct RX listening sound inside Mic App -> works 
    and correct audio in other receiver app after using Mic app. (your original problem description) -> works

Very nice! Thank you very much for your fast support!

If i could help anyway with my hardware till you get yours, let me know.

miotislucifugis, 2022-05-05:
Seems to be working for me too.  H2 with CPLD QFP100.  Great work!

sivaelid, 2022-05-05:
Hi have tried the latest BIN  and tried microphone transmission to a HAM RadioTransceiver  and confirm both audio and CTCSS and Roger bleep all working great job. Simon

Brumi-2021, 2022-05-05:
> I haven't looked at to see which platform or clpd I have as I have 5 of them and they all behaved the same until I went back to 1.4.3 then everything was working again except for ctcss and roger beep. I've been in using the tx app for tx for now and haven't tested rx.

@deano817 , we would appreciate if you also can check the test version that uploaded here . 
thanks

Brumi-2021, 2022-05-05:
> I also have this issue with any version above 1.4.3 with WM audio chip H2

Hi , @TheDaylightWalker , we would also appreciate that you test the test-version that I uploaded here .
Thanks ,

deano817, 2022-05-05:
> 

I can't get it to flash at all.

diphtong, 2022-05-06:
I flashed with Linux, works without problems.
Flashing on Windows doesnt work for me either...dont know why.

Brumi-2021, 2022-05-06:
Hi @deano817 , oh i see . But where is the problem ? You can not  put your device in Hackrf mode,  with blue screen ? 
I normally use linux, but in the past in some windows update , I also need to reinstall zadig .
Cheers

diphtong, 2022-05-06:
i had also problems flashing in the past and found the USB cable was the problem. Maybe check with another USB cable.

TheDaylightWalker, 2022-05-06:
On windows you need to use driver which comes in official zip file, because it's USB composite device and other drivers don't work properly. Also no need for zadig libusb downgrade. This driver works with all sdr software I've tried, other drivers won't work on some specific sdr software

deano817, 2022-05-06:
Got it to work and I can confirm that ctcss and roger beep are working and Rx is normal. Thank you.

Brumi-2021, 2022-05-07:
Hi ,  @ArjanOnwezen , 
now that it is confirmed and merged , whenever you can , this issue can also be closed .
🙏🏻

ArjanOnwezen, 2022-05-07:
Really nice work @Brumi-2021 !

closed 2022-05-07 by ArjanOnwezen