portapack-mayhem/mayhem-firmware #622: No Audio or white noise

jLynx, 2022-05-03:
**Describe the bug**
This is a ticket to track the progress of this bug where either there is no sound or only white noise coming from the device.

If you have this issue, please comment the device you have (https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions) and the version of firmware you are currently running.

We have been working on trying to get to the bottom of this bug but its not a simple fix sadly. That means there is no ETA for when this will be fixed

jLynx, 2022-05-03:
[image]
Just adding my chat log for reference on where we are at

jLynx, 2022-05-03:
If you are having this bug can you please try boot with every option listed here https://github.com/eried/portapack-mayhem/wiki/Won't-boot as these may fix your sound issue. If it works please comment here letting me know so

ytsqkl, 2022-05-03:
i  try  20 times，sucessed only one time，when restart it ，failed again.

ArjanOnwezen, 2022-05-03:
> i try 20 times，sucessed only one time，when restart it ，failed again.

What device do you have and did you remove coin battery?

ArjanOnwezen, 2022-05-03:
I have exactly the same issue as described above

[image]
[image]
Above settings are when it's not working

ArjanOnwezen, 2022-05-03:
These are the layouts, as also described by other user.

`working

00 > 40 00 06 40 44 53 0B 14
08 > 00 00 60 00 E1 E1 E1 00
10 > 80 80 80 6E 18 00 00 00
18 > 00 00 0C 01 00 02 B0 1F
20 > 9F 20 00 00 00 00 00 00
28 > 00 00 00 00 00 00 00 00
30 > 00 C1 00 00 00 00 00 00
38 > 00 00 00 00 00 00 00 00
40 > 00 00 00 00 00 00 00 00
48 > 00 00 00 00 00 00 00 00

not working (this is most of the time)

00 > 40 00 06 00 44 52 0B 14
08 > 00 00 60 00 E1 E1 E1 00
10 > 80 80 00 6E 18 00 00 00
18 > 00 00 0C 01 00 03 B0 1F
20 > 9F 20 00 00 00 00 00 00
28 > 00 00 00 00 00 00 00 00
30 > 00 C1 00 00 00 00 00 00
38 > 00 00 00 00 00 00 00 00
40 > 00 00 00 00 00 00 00 00
48 > 00 00 00 00 00 00 00 00

ytsqkl, 2022-05-03:
> 

yes,remove 1 minute,  h2  r2

kolondrej, 2022-05-03:
Hello. I have version H1 R2. I took the battery out for a minute. And tried all the boot options. Assembly version Firmware Version: da49743c (API:1.04). There is white noise. The AK4951 dump is the same as above (not working). After downloading the firmware from HAVOC Firmware Version: local-ade1b976 (API: 1.03)
Sound is present. On the firmware from HAVOC I can not check the dump.

ytsqkl, 2022-05-04:
Thank you very much for your contribution, ham all over the world are very happy, I have a suggestion, you can first return the program to version 1.4.4, make sure that the sound decoding is normal, and then compare the next version that cannot decode the sound to see about the analog sound decoding What went wrong, then fix it.

jLynx, 2022-05-04:
> Thank you very much for your contribution, ham all over the world are very happy, I have a suggestion, you can first return the program to version 1.4.4, make sure that the sound decoding is normal, [...]

Sadly it's not as simple as that as it's a timing/clock issue. If we revert then we end up braking compatibility to a whole bunch of portapack devices. Right now it's a trade off till we can get sound working for you. Keep in mind you  lot are the minority. Sound it working for most people, it sonly a handful that are having this issue. For example I am not having this issue which makes it extremely difficult to debug as I need to rely on making changes and having you lot test it for me and tell me the outcome.

jLynx, 2022-05-04:
I think the first step we need to do here is what @sharebrained said and to add some sort of simple checksum to persistent memory, to detect if the contents were bad, and initialize them if they were. It would be maybe a CRC16 or CRC32 calculated on a copy of the data in regular RAM. The persistent memory would be copied to regular RAM at startup, and read or written from there. Every second (or 100 milliseconds, or whatever seems appropriate), the copy in regular RAM would be locked (to prevent changes), the CRC calculated and appended, and the result written to persistent memory.

I don't have a lot of time on my hands, so if anyone wanted to help out with developing this, please let me know. I will do as much as I can in my free time

jLynx, 2022-05-04:
Can anyone having this issue please test this firmware and let me know if it fixed the issue?
[portapack-h1_h2-mayhem_aabaa35.zip](https://github.com/eried/portapack-mayhem/files/8617578/portapack-h1_h2-mayhem_aabaa35.zip)

ytsqkl, 2022-05-04:
i will test  ,tks

ytsqkl, 2022-05-04:
oh  my god  you fix it，analog audio work normal

jLynx, 2022-05-04:
We just need more people to test it to confirm. As I have done similar fixes before that fixed it for some people but not others as this issue is not very straightforward sadly

ytsqkl, 2022-05-04:
Looking forward to bei completely repaired

kolondrej, 2022-05-04:
> Can anyone having this issue please test this firmware and let me know if it fixed the issue? [portapack-h1_h2-mayhem_aabaa35.zip](https://github.com/eried/portapack-mayhem/files/8617578/portapack-h1_h2-mayhem_aabaa35.zip)

Version H1 R2. Audio Rx WFM is worked.

ArjanOnwezen, 2022-05-04:
> Can anyone having this issue please test this firmware and let me know if it fixed the issue? [portapack-h1_h2-mayhem_aabaa35.zip](https://github.com/eried/portapack-mayhem/files/8617578/portapack-h1_h2-mayhem_aabaa35.zip)

Thanks @jLynx that firmware version solved the issue.

jLynx, 2022-05-04:
Awesome. I sort of just chucked delays at it and hoped one/some of them would solve the issue. I am thinking we merge it in now (As 3 successful tests) and see if everyone else with the issue stops complaining. If that works then I can work on removing some of the potentially unnecessary delays

jLynx, 2022-05-04:
I will merge it tomorrow my time. Once merged I will also keep this issue open for a week or so to stop duplicates been created of people using 1.5.1 and not using the search function on GitHub

ArjanOnwezen, 2022-05-04:
> I will merge it tomorrow my time. Once merged I will also keep this issue open for a week or so to stop duplicates been created of people using 1.5.1 and not using the search function on GitHub

Great! You can merge it any time. It improves it for at least for three cases and has no side effects.

rizal3204, 2022-05-04:
@jLynx  that firmware solved the issue thanks btw im using h1 r1 ak4951

GullCode, 2022-05-04:
@Jlynx I'll make a test in the morning too

aledantas2k12, 2022-05-04:
> Can anyone having this issue please test this firmware and let me know if it fixed the issue? [portapack-h1_h2-mayhem_aabaa35.zip](https://github.com/eried/portapack-mayhem/files/8617578/portapack-h1_h2-mayhem_aabaa35.zip)
@jLynx 
Sweeet. It's working now.

GullCode, 2022-05-04:
@jLynx @ArjanOnwezen I can confirm this is working and decoding correctly on H1 AK and H2 WM
Tested on local FM station @ 92.00MHz and AM DSB on local airport @ 126,175 MHz, working fine on both with that firmware.

GullCode, 2022-05-04:
Looking at the modifications, it looks like it's kind of sd card driver related. 
We have to give more time because it seems to load slower on some archs.

ArjanOnwezen, 2022-05-04:
> Looking at the modifications, it looks like it's kind of sd card driver related. We have to give more time because it seems to load slower on some archs.

I prefer loading this hotfix.
Currently many devices aren't working correctly.
In parallel you could figure out the root cause.

GullCode, 2022-05-04:
I was talking about time in the code @ArjanOnwezen X-D

jboone, 2022-05-04:
> [image]
> Just adding my chat log for reference on where we are at

Adding my prior comments when talking with @jLynx as text to be more searchable:

> So the perplexing thing here is that I’ve shipped PortaPacks with both the WM8731 (2014-2017 or so) and AK4951EN (2017-2022). I had detection code back in 2017 that would identify the audio chip at boot time and do the required initialization, then manage the audio chip through a driver abstraction.
> So one big problem with using the LC43xx persistent memory is that there’s no way to detect if the contents are random or otherwise invalid. I think a lot of start-up issues people have had are the result of having garbage in the persistent memory, causing code that doesn’t check for valid values to crash or otherwise act erratically. This issue was magnified as more and more applications were added that depended on that tiny block of persistent memory.
> I wanted to add some sort of simple checksum to persistent memory, to detect if the contents were bad, and initialize them if they were. But I never got to it, and now don’t have the time or inclination to implement it.
> But in a nutshell, it would be maybe a CRC16 or CRC32 calculated on a *copy* of the data in regular RAM. The persistent memory would be copied to regular RAM at startup, and read or written from there. Every second (or 100 milliseconds, or whatever seems appropriate), the copy in regular RAM would be locked (to prevent changes), the CRC calculated and appended, and the result written to persistent memory.
> Do you have a sense for whether the white noise issue affected *all* the PortaPack variants, or just ones with a specific audio chip or other specific hardware characteristic?
> So there’s three places where 1ms sleeps were added. Are all three required to address the white noise issue, or just the one with the comment saying so. What happens if the other two are removed?
> I’m hard-pressed to guess why the sleep matters. My first inclination is like yours, in that there’s a timing issue. Perhaps the I2C peripheral talking to the audio codec isn’t quite ready yet. Though that seems unlikely, as the I2C peripheral is initialized long before. Or maybe there’s some side-effect of the CPLD check/program process that is interfering with the audio codec initialization. It’s really hard to tell, short of spending some hours with a unit that’s experiencing the problem, poking at it with an oscilloscope or logic analyzer.

jLynx, 2022-05-04:
> I was talking about time in the code @ArjanOnwezen X-D

I changed it from 10 to 100ms in the places that matter the most, so it should be fine as is. If we get users still complaining after we have merged this, then we can look at increasing it further.

zigad, 2022-05-04:
> Can anyone having this issue please test this firmware and let me know if it fixed the issue?
> [portapack-h1_h2-mayhem_aabaa35.zip](https://github.com/eried/portapack-mayhem/files/8617578/portapack-h1_h2-mayhem_aabaa35.zip)

Hi @jLynx I can confirm it also works now on my [Portapack H2+R1](https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions#h2r1) with this firmware

jLynx, 2022-05-04:
Leaving this open for another week or so to make sure it's no longer an issue. I have merged the code in so it will get released in the nightly build in a few hours time

jLynx, 2022-05-05:
Please test out the latest nightly release https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2022-05-05

rkalasek1, 2022-05-05:
this nightly works great issue is gone  no holding down button during boot to get audio thanks

zigad, 2022-05-06:
I also tried latest nightly release `n_220506` and it works as expected. No need to push any buttons or remove SD Card

kolondrej, 2022-05-06:
H1R2 Nightly release - 2022-05-05 don't work. a'm have used all of metod of reset (Without dispatch of batery). aabaa35 working.

artyfarty, 2022-05-07:
Nightly build fixed a problem for me, just flashed a new version and voila.
None of initial workarounds did.

kolondrej, 2022-05-07:
Version H1 R2. Audio Rx WFM is worked on 2022-05-07.

ArjanOnwezen, 2022-05-11:
@jLynx Great job! You made many people very happy... Closing this issue.

closed 2022-05-11 by ArjanOnwezen

RomixERR, 2022-05-11:
Раньше на v1.5.1 звука на WFM не было, теперь (n_220509) всё появилось. Спасибо!