portapack-mayhem/mayhem-firmware #624: Problem upgrading the firmware

adammanella, 2022-05-03:
**What is happening?**
Describe here why you are unable to upgrade the firmware. Before describing your problems, **do the following**: 
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

....black screen on H2 after 1.5.1 update... 

hackrf_info shows the following:

adammanella@Adams-MBP mayhem_v1 % hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc365798c3
Board ID Number: 2 (HackRF One)
Firmware Version: v1.5.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x00614774

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

it was shipped with version 1.4.0 but was constantly freezing and unable to enter hackrf mode

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
Add photos and videos of your procedure.

ArjanOnwezen, 2022-05-03:
Please try https://github.com/eried/portapack-mayhem/wiki/Won't-boot first

adammanella, 2022-05-03:
THANK YOU!!!!! IM SO HAPPY!!!

ArjanOnwezen, 2022-05-03:
Solved

closed 2022-05-03 by ArjanOnwezen