portapack-mayhem/mayhem-firmware #635: Problem upgrading the firmware

zimsch0, 2022-05-07:
**What is happening?**
Describe here why you are unable to upgrade the firmware. Before describing your problems, **do the following**: 
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
Add photos and videos of your procedure.

ArjanOnwezen, 2022-05-07:
@zimsch0  Please add description of your issue and what you've tried so far.

jLynx, 2022-05-07:
Closed due to invalid issue.

closed 2022-05-07 by jLynx