portapack-mayhem/mayhem-firmware #636: Freeze when opening Capture app immediately after boot

cygeus, 2022-05-07:
**Bug description**
The GUI freezes when one opens the Capture app immediately after boot.

**Steps to reproduce**
1. Switch on the PortaPack.
2. Turn the encoder 3 times to the right to select the 'Capture' icon.
3. Press the Enter button to start the Capture app.

**Actual behavior**
The GUI freezes. The Main Menu is still displayed with the Capture icon highlighted. The clock in the bottom right is not updated anymore.
Turning the encoder wheel has no effect. Pressing any of the buttons has no effect either. The only way out is to power off the device and start it again.

**Expected behavior**
The Capture app opens.

**Affected versions**
* Firmware: I only tested using the latest nightly build: n_220507 (can test other builds if needed)
* Hardware: HackRF One + PortaPack H2+R3 (both Chinese clones)

**Additional: workaround (iow alternative way to open the Capture app, causing no freeze)**
1. Switch on the PortaPack.
2. Press the Enter button to start the Receive app.
3. Press the Enter button to start the ADS-B receiver.
4. Use the buttons to return to the Main Menu.
5. Turn the encoder 3 times to the right to select the 'Capture' icon.
6. Press the Enter button to start the Capture app.
7. The Capture app starts and one can see the waterfall display scrolling.

Brumi-2021, 2022-05-07:
Hi @cygeus , I still did not reconfirm the build n_220507,
I will do later, with H1 AK4951. But  I believe  it should be ok behaviour. 
But how about in your set up,  using  previous build, or fw 1.5 1 or fw 1.4.3 ?  Thanks .

Brumi-2021, 2022-05-07:
Hi, I checked n_220507 , in my set up , H1-AK4951-CPLD QFP60, and as I expected,   everything smoothly with Capture App., no problem at all ,   (we can change BW= rate,  gain , amp, .freq, , step, and record correct C16 files,....)  

Pls. give us more details , and make sure if it is specific problem of a fw version , and from which version was OK, , or  some other HW problem ???

cygeus, 2022-05-07:
I'm able to reproduce this issue with v1.51 and v1.5.0 from this repo.
And with v1.4.3, v1.4.2 and v1.4.0 from the jumbo77/portapackh2 repo. (Due to the QFP100 being present in my H2+R3.)

Brumi-2021, 2022-05-08:
Hi , but then , 
all the FW versions that you checked was showing the same problem ?  
you did not find any FW version working well , even the original fw that was installed when you bought it ? 
I mean , it is not a FW degradation , you always have that problem ? 
did you buy the unit all together assembled Hackrf + Portapack H2R3 , or did you buy it from different vendors , and you assy them ? 

How about other H2R3 users  (means H2/WM/CPLD QFP100) ?

cygeus, 2022-05-08:
I'm totally confused now. While trying to pinpoint the problem, I flashed multiple different versions of the firmware and all versions had the same issue. I read parts of the source code, build the firmware myself from scratch.
And then suddenly I could not reproduce the problem anymore. Any nightly build or release from this repo or from jumbo77 - which previously all failed to open the Capture app - just worked fine. Unfortunately it is unclear to me what caused the issue and what solved it.
At one point I did a `hackrf_spiflash -c` to clear the SPI flash, but I'm not 100% sure if that solved it.
And then, after some more flashing, the problem occured again with nightly build 220507. I'm a bit out of ideas for now, will dig some more later.

To answer your questions:
- I did buy the HackRF and PortaPack separately, so there was no mayhem firmware preinstalled on the HackRF.
- If this problem occurs, transmitting works just fine. I did send my callsign using CW and APRS and could receive and decode both correctly using a RTLSDR dongle.

Brumi-2021, 2022-05-08:
I see , glad that you could recover once . But not so happy if it is still unclear about the process to recover it.  After reading your comments , as you said , all it is so confusing to me ,  and I am not  any expert about CPLD issues. 
Maybe other more advanced user may help you. 
@jLynx any clue about it ?

jLynx, 2022-05-08:
I'm pretty sure it's do do with corrupt memory which is an issue we have been fighting for a long time as we do not validate the memory. It's working for you now. But if it wasn't I would say try remove the small rtc battery and test if that fixes it. Then you can add it back in. This will clear the persistent memory

ArjanOnwezen, 2022-05-11:
closing issue, since issue mentioned is solved.