portapack-mayhem/mayhem-firmware #637: Shadow signal appears after TX'ing in any portapack App

TheDaylightWalker, 2022-05-07:
**Describe the bug**
Shadow signal (of ~1.7MHz bellow) appears on receiving app after using any App, where you press TX button. 

**To Reproduce**
Steps to reproduce the behavior:
Go to Receive, Audio, set WFM Broadcast station frequency (just because it's easy to test, and you get shadow signal easy, but it doesn't depend on NFM/WFM/AM/Special)
Go to any transmitting app, and try to transmit for second or two. 
Then go back to Audio, to the same frequency you tried to listen before, you will encounter signal got worse, and you will see shadow signal on the waterfall. Signal is from the station which is 1,7MHz below 
**Expected behavior**
Receiving signal should be as good as it was before using TX'ing app without shadow signal  on waterfall and no noise on center receiving frequency. 

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
All versions from 1.4.3 including have this issue for 100%
Tried with older 1.3.1, so after TX'ing, and getting back to Audio, device simply hangs, and RX led is blinking gast 

**Additional**
My device is H2+ with WM8731, pcb v3.4B

Brumi-2021, 2022-05-07:
Hi thanks for finding and reporting it .
we have been discussing for a while about it in Discord .  (initially considering many things , including PCB shielding quality ,  battery switching noises .... ) 
Initially I could not replicate about it , because I was checking also in 106 Mhz , (but that area is quite empty in my local area , 
and  you need to check it in a crowded FM band  frequency area.

The problem is clear ,  and at the moment confirmed in two receiver App , but maybe it affects to more receivers (not checked) 

As @TheDaylightWalker reported , 
1-) Ex , **the receiver FM  Station , appears , much clean and with better S/N  , if we tune it directly after boot ,  than if we used before some Transmitter TX  App.**   (And that is noticeable in  Audio receiver app and also in the Capture appl)

Example  good signal  , (Audio App) WFM  demod, just  after reset booting : 
[image]

Same FM station , same conditions (same antenna , same Hackrf postion, same GAIN , RF AMP LNA , ...) but 
just when we used the  TX  key in any transmitter App before ,
[image]

Also similar  things happens , and maybe more visible , when using  Capture app (just using  waterfall , not recording ) 
Just  after reset booting , Capture app --> quite clean  signal , depending on the anntena ....
[image]

But if we go before to any Transmitter app and press TX  , --> later when going back to Capture app -->  quite ghost noise signals sourroiunding , degradating  the S/N .
[image]

I already confirmed it , using  latest todays  fw . night n_220507 
**But  the good thing , is that in the fw 1.3.1 , we did not have that problem !!!**  ( always clean station, same as boot conditions) .

(at first glance ,  it  seems , bad   IF BPF  BW  register settings , that produces more incorrect  mixed beats, )  
Pending to check ,  from  which fw version , we introduced  that degradation, find the root cause , and fix it  .
 
To be  continued , 

If you can also reproduce it  ,  let us know ,

Brumi-2021, 2022-05-10:
Hello @TheDaylightWalker 
I could confirm you that this problem was NOT there , in the fw version 1,3,1 and 1.4.0  
**But from the  fw version 1.4.1 ONWARDS,  we have introduced that bug or side effect problem .**

It is not a serious problem ,  but it introduces  those "ghost or beat noise" that reduces the S/N  of the received signals,
when we have some adjacent channel , with  strong signal  stations  +/- 1.7 Mhz aprox

In next days, i will deeply  study , checking  all  the delta commit changes that we applied from 1.4.0 --> 1.4.1 
and trying to fix ,  and recover the good RX  reception that we had . 

(Left side ,  clear RX from fresh reset  booting /  right part , after using some  Transmitter TX and go back to the same receiver.
The beat noises , can appear below or upper the tuned frequency , depending on the srong adjacent channel position.

[image]

[image]

GullCode, 2022-05-10:
If no one sorted this tomorrow, I'll git bissect.
If some have the time right now, go and bissect ! (https://git-scm.com/docs/git-bisect)

Brumi-2021, 2022-05-11:
Hi @GullCode , did  you  already reproduce it,
in your local area ?   (I guess yes , if you have many strong FM stations and crowded spectrum)

If  nobody fix it before , I will start the investigation from next week end .
 Any progress about it , let's  keep reporting here .
Cheers

Brumi-2021, 2022-05-13:
Hi,  after bisecting that problem introduced in fw 1.4.1  ,
 I could find the small pcs of code that was deleted in that commit , #365 ,  and produced that problem 

[image]

After  adding back those lines in portapack-mayhem/firmware/application/radio.cpp
 , that particular  problem (ghost  beats in RX  from adjacent strong signal channels , when comming back from the TX )   dissappear in my platform H1 , AK4951 ,  (good) , but the side effect , is that in my AK platform ,  the switching time between RX /TX inside Mic App has some small delay . 

But in the H2 WM8731 platform , 
we got again the intermitent switching @Teixeluis solved ... (RX led is blinking fast and device hangs) 
Then it is not good solution  for all (WM - AK )  ,  we need to investigate more about it .
Any help is welcome , 
Cheers,

Brumi-2021, 2022-05-22:
Hi , @TheDaylightWalker 
I could  also add the same c/m in both points, (covering all  Transmitters exits ) : 
1-) Exiting  Mic App ,
2-) Going back from the Transmitter group,  from  any of both options 
a) Normal way , pressing (left arrow key in top title bar)
b-) advanced user way , "big left arrow Icon" (in case that user have activated in UI Settings "add return ICON in GUI" 

I confirm in both platforms H1 AK / H2+ WM -QFP100 , and in both the problem is gone. 
Waiting for your additional  validation , for safer release PR. 

By the way , @TheDaylightWalker , if you still have  some random latch ups in your H2+ WM QFP100, I suggest you to update your Hackrf fw before flashing the Mayhem ,  (in my case , all those issues dissappeared at the moment ) .

Cheers
[portapack-h1_h2-mayhem_637_final.zip](https://github.com/eried/portapack-mayhem/files/8749884/portapack-h1_h2-mayhem_637_final.zip)

TheDaylightWalker, 2022-05-23:
Hi. My portapack is QFP64 with WM8731.  With this 637_final I have constant glitches in audio app simply exiting and entering app again (simply stucks until powercycle, values of frequency and gain are zeroed and waterfall black). I've reflashed hrfone standalone, but issues with device hanging and etc haven't gone. Shadow signal seem to disappear if I'm able to get to any TX'ing app, to TX and back to Audio app 1st time, but issue reappears if I repeat those steps several times. But it mostly hangs with this version. 
1.5.3 is very stable and I don't encounter those issues despite ghost signal. Maybe there are more users with H2, QFP64 and WM who could try to test and feedback.

Brumi-2021, 2022-05-29:
Hi , summarising my current understanding 

1-) that problem (RX S/N  degradation , after using any transmitter App,  compared to the starting  RX S/N from booting ) happens in any  platform  ,  H1 AK4951 / H2+ WM8731 ,...

2-) I confirmed that although the problem ,  it is clearly visible on the LCD waterfall screen  of the  RX Capture app,/ Audio App.  , the problem, seems to affect to all receivers. 
Example ,  I could confirm , that inside my home , just booting -fresh reset reception -  I easily can capture  3 or 4 airplanes in ADSB , but  when using  ADSB rx after using any  transmitter app , --> I got 0 airplanes (with same set up ) . 

3-) It seems , as i said  above , that problems was not there  in fw <= 1.4.0,   it was introduced from fw 1.4.1  onwards.

4-) Based on above git bisecting data  1.4.0 --> 1.4.1.  the problem started to appear , after that commit that solved  a big  unstability in the WM Platform ,
[image]

5-) If  we added back that  pcs of code  in the same part as it was, 
and we really  can solve the problem for that Mic App tx on AK platforms   , but we produced many different hang ups on H2+ WM Platform (as Teixeluis find out ) .
 
6-) Recently , I  managed to modify slightly that code , and add it  (hackrf::cpld::load_sram() ) ) 
 example , at  the exit of  Mic App , and  it avoids the hang up , and it works 100% on  AK platform , 
 but in H2+ WM Platorm (there is no clear solution now ) 
 it seems to work at 50% . (pending to double check with my h2+ unit )  , and adding the same code in some other Tx app sometimes it produces hang ups ... 

So apparently, just from experimental point of view,   the  problem is  related to that  Hackrf CPLD  load from eprom or sram.
But stil not understanding it ...

We (@TheDaylightWalker and me )   think that  the problem is  there since  long time ago ,  and we just did not realised it before ...  (because it is easy to detect using  waterall receiver applications in areas , where you may have many strong adjacent channels . ) 

We need to investigate more , till solving it .

Cheers.

TheDaylightWalker, 2022-07-12:
any progress ?

Brumi-2021, 2022-07-17:
Sorry , not yet.   Recently I have been   busy with the the big Portapack CPLD TQFP100 , trying to read and display its IDCODE in a new Debug App , (but still  not solved, but postponed ) and this week, trying to update my linux laptop linux ... soon I hope to fix it and continue with the investigation.

Brumi-2021, 2022-08-26:
Hi , @TheDaylightWalker  , (and all) 
recovering those investigations, let me share with you my latest progress about it , 

I could confirm , that using Mic Tx transmission,  and then going  to any audio RX , all the reception band has some  spurious ghost signals that creates  interference and background noise to all band  . 

I though that it was due to strong adjacent channels ,  but now  , I  do not think so  ...  My current thought is that any strong signal (main) can generate a ghost , if you use prior any Transmitter App, and later you look for it (with the proper delta shift )  .
(if you do not use any TX App after the boot , all the RX  are clean without any ghost signal  - that is the good RX reference) 

Example table of my local main WFM signals 
, Using Mic App, --> the created ghost signal is a clone of the main (now I know  becasue it has the same audio contents ) , and alwasys is created shifted from main , around +1,55 Mhz , (but using other TX App, the shifted ghost freq, might change slighly , ex +1.7 Mhz 
[image]

I measured the RX  current consumption with / without  ghost signal , looking for some diference.
(I was assuming that we switch on some clock or  some  IF settings in the  TX , that we did not switch off in RX time) 
But  there was no signifcant current mA difference ! 

I tried many other sw   c/m's (radio.disable , cpld re-init , ...)  ,  but none of them worked beside the same may-june's c/m  .
the only  good  c/m that i found 100% effective is to call again load_sram() for the Hackrf Xilinx CPLD (CoolRunner II) .

I am not at all a CPLD specialist (then i might be wrong , pls correct me if needed ) 
I tried to read  a lot of technical documents about the CoolRunner , and CPLD's (XC2C64A CoolRunner-II) ,  and I understood that our  Xilinx CoolRunner II CPLD  , **has a Hybrid memory Architecture ,**  

 Hybrid SRAM devices that have an SRAM FPGA architecture and a nonvolatile configuration memory on-chip. These Hybrid FPGAs must be loaded internally on each power- up cycle. Only after the device is loaded with the configuration, can it start operating according to the customer application. After each power-down or power supply "brownout," the Hybrid device loses its configuration and needs to be loaded again in the next power-up cycle.

[image]

[image]

My current theory , is that maybe when we use alternatively the TX / RX paths ,  maybe some low level driver  part code of the Hackrf sumodule is maybe handling that  CPLD sram refresh dinamically ???  , and leave at the exit of the TX App ,  in a wrong  programmable switches status our CPLD .therefore , the c/m  looks effective.  (just  my current theory )  . But  I do not know  , why this at the end , this might produce a beat with the main signal ???
[image]

Anyway , at the moment , I added that effective  c/m to only the Mic App  (as starting test) ,
 (so, now ,  that TX App should not generate any ghost signal in RX now )
Through discord channel ,  we will follow the testing , and if no side effects, we will expand them to the rest of TX App.s

Maybe some experienced original Portapack developpers may clarify us  , about those lines , that I also found  inside the code ,
        // Use CPLD's EEPROM config when transmitting
	// Use the SRAM overlay when receiving

But somehow, in line with the countermeasure ,...
Any further help , or comment will be appreciated, 

That's all by now  ! 
Cheers.

Brumi-2021, 2022-11-22:
Hi  , after those two above merges,
I could not reproduce that problem anymore . 
So , in my opinion ,  it can be close it .

gullradriel, 2022-11-22:
That was a good one. Closing !

closed 2022-11-22 by gullradriel