portapack-mayhem/mayhem-firmware #638: Issues upgrading Firmware

wr417h01, 2022-05-07:
Error message when attempting to update the firmware on my HackRF One PortaPack.

Windows 10 machine

I am attempting to upgrade the Firmware on my Hackrf One PortaPack H1  to the Mayhem firmware. I formatted the micro SSD and extracted the firmware files to a folder on my desktop. I ensured the Micro USB cable I am using allows Microsoft Windows to recognize the device. In device manager I see HackRF One under Universal Serial Bus Devices. 

My issues is when selecting the flash_portapack_mayhem.bat file I get an error stating "The code execution cannot proceed because MSVRC120.dll was not found. Reinstalling the program may fix this problem."

I receive this error directly after the prompt to "Press and key to continue . . ."

I have tried using the dfu fix, I hack checked the HackRF One drivers in device manager and am still having the same issue.

I tried using Zadig to update the USB driver.

I am at a loss and was hoping someone could point me in the right direction.

Thanks. 
[image]

wr417h01, 2022-05-07:
Also, When trying to manually update the HackRF One driver I get this error.

[image]

TheDaylightWalker, 2022-05-08:
last fw zip included driver was without needed folder (driver/amd64), get an older firmware pack, it's driver folder contains full install
No need for zadig, just use driver for example from 1.4.3 and it should start working properly

eried, 2022-05-09:
@wr417h01 so, was it solved? (I saw the message in discord)

ArjanOnwezen, 2022-05-12:
@wr417h01  Please let us know if issue is resolved.