portapack-mayhem/mayhem-firmware #639: Rec button failure for NRF and BLE

falindrknes, 2022-05-08:
The, "REC" button caused the unit to freeze up completely. 
Only hard restarts can make the unit functional again.

- Power up unit
-Go to NRF or BLE
Press the record button.

Unit will fully freeze, screen will freeze, 
Operations will freeze.

Jumbos Firmware as well as Mayhem 151 (stable)
Have been proven to lead to the same "freeze".

falindrknes, 2022-05-08:
I did see a minor solution before where a dev removed the record button all together and I think the software auto recorded. I think it was a modified version of 1.3.x;
Just hoping a fix is possible; latest firmware os amazing.

GullCode, 2022-05-08:
I can confirm the bug on my H1R2, even on the last nightly.

loewal, 2022-05-08:
Yes, even in AFSK.
In APRS it's ok.

Brumi-2021, 2022-05-08:
Sorry I never used the REC button in those functions . But probably that bug or incomplete function, has been always there , not ? 
I can also reproduce it even the old  FW 1.3.1. , in my H1/R2 (AK and QFP64 CPLD)

jLynx, 2022-05-08:
In that case I guess we should remove the record button for now if the feature is incomplete. What's everyone's thoughts on this?

loewal, 2022-05-08:
My idea 👍

GullCode, 2022-05-08:
Isn't it recording by default?

Brumi-2021, 2022-05-08:
Hi I do not know about the completion level or previous history of this REC functionality in those Apps.

But all of us , are quite sensitive to the potential regressions of each new release .
 Therefore we should better distinguish about generic label “bug” :
 * new bug found from the new feature under dev .
 * existing bug that was always there ,  (and just  not detected before)
 * or bug coming from a pure regression side effect when adding new feature as a side effect , and we break or damage and old good feature, because that old feature was working well in previous releases  ) .  
 
The problem is that when open an issue bug , normally we do not know in which case are we , till we start to investigate it ...

In that sense , coming back to your @jLynx question ,  if that bug or incomplete job was there from many version before , maybe better to note as “new feature to be finalised” and now just remove the button (if it just freeze the App) . I agree with you .
Just my opinion

ArjanOnwezen, 2022-05-09:
Duplicate of #387

closed 2022-05-09 by ArjanOnwezen