portapack-mayhem/mayhem-firmware #641: Problem upgrading the firmware

TQMatvey, 2022-05-09:
Hello, while i was building my own builds, and flashing i got i/o error (probably i accidentally took cable out while flashing), it booted but now i am unable to go to hackrf mode (black screen and device not detected, tried ubuntu 22.04 and windows 11 with all drivers) also flashed dfu stuff, device detected in device mode but as dfu (firmware not flashable as hackrf is not detected), i watched vids and did all troubleshooting stuff....

Brumi-2021, 2022-05-09:
Hello i also had problem once  (same as you probably bad cable contact while flashing ) . Later I will give you details how to solve it

Brumi-2021, 2022-05-09:
Hello, let me explain , my understanding .
Most likely you have corrupted your flash memory , and damage the program part  that handle the hackrf mode --> usual blue screen . And as you can not put it in hackrf-mode,  you can not reflash it .  

In this conditions  while you are trying to activate the hackrf-mode, you just got black screen , 
and no USB activity at all.  (we need to recover it ! )

But do not worry , you still have DFU !!!!!!

but just follow those steps :  
1-) go to the exact  release package.zip that you were using when you had that problem.  (download the full release package)
2-) extract all the files of that package . (you will get something like this picture, it is just example)
(the important files ,  is the those blue ones,) 

[image]

3) Put your  hackrf in  DFU  mode (while holding push DFU, press RESET ) --> you will get black screen.

4-) send from the PC terminal that  command  : 
**dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset** 
    
(in that DFU mode, all hack-rf answer with the same device id  1fc9:00c .)
and after sending that command , you will see the progress on your pc , 
[image]

once finished the process ---> your hackrf will be  in hackrf-mode  (normal blue screen ) --> and from that point , 
you will be able to reflash it  with the mayhem binary . 

I hope it will also work to your device. Sure!!!! 
Let us know , to close the issue . 

Note :  I could learnt it , from  that Greatscottgadgets link  ,  really very usefull .
https://github.com/greatscottgadgets/hackrf/issues/835

TQMatvey, 2022-05-09:
Hello, i Tried your steps but i am not getting blue screen and my hackrf isnt being detected after i flash it.. 
[image]

TQMatvey, 2022-05-09:
oh an trying to run 
`➜  hac hackrf_spiflash -w portapack-h1_h2-mayhem.bin                                                  
File size 936836 bytes.
hackrf_open() failed: HackRF not found (-5)`
well returns an error always

TQMatvey, 2022-05-10:
and i just noticed that barely anything opens, trying to open record/replay menu freezes my hackrf, but some stuff like ads-b receveiver work fine

TQMatvey, 2022-05-10:
I tried all steps with different cables, linuxes, windowses 10,11, laptop, pc

Brumi-2021, 2022-05-10:
Yes your sceens seems correct , same as my notes. 
But when I compared with with Eried's DFU advices , maybe I missed to tell you step 5 (reboot)  , I am not sure , pls try 
[image]

TQMatvey, 2022-05-10:
sadly this doesnt change anything, it never shows up in lsusb/device manager(win) and trying to flash normal hackrf firmware returns that no hackrf is found

Brumi-2021, 2022-05-10:
But when you put it in DFU mode ,at that time ,  you can see the answer of your device , using lsbusb, no ?

TQMatvey, 2022-05-10:
I tried and reb9otin, i flashed dfu like 20 times by now but it 

> But when you put it in DFU mode ,at that time , you can see the answer of your device , using lsbusb, no ?

Yes and i can flash hackrf_one_usb_dfu also

Brumi-2021, 2022-05-10:
I recheck my notes again. In my case I need to repeat the process two times , because that reboot did not work at all  . Pls repeat again from a correct DFU and you should get eventually blue screen , no ?

TQMatvey, 2022-05-10:
nop... i tried everything by now..

Brumi-2021, 2022-05-10:
While you are in DFU , what you get from the command lsbusb?

TQMatvey, 2022-05-10:
> While you are in DFU , what you get from the command lsbusb?

`Bus 003 Device 005: ID 1fc9:000c NXP Semiconductors LPC4330FET180 [ARM Cortex M4 + M0] (device firmware upgrade mode)`

[image]

Brumi-2021, 2022-05-10:
It is fine. The DFU boot loader is answering you 
Correctly . Then it should work 🤔

TQMatvey, 2022-05-10:
but it doesnt lmao thats the issue, i never get to see hackrfone in lsusb or get a blue screen on hackrf itself....

Brumi-2021, 2022-05-10:
But then , maybe you are now in  another strange situation , that I also had . 

Beside the previous DFU problem , that i explained you how i recovered . 

Another  day, suddenly from linux I started to have that strange msg , from spi flash tool
hackrf_open() failed: HackRF not found (-5) . 

And when I tried to put in Hackrf mode  from linux , not visible  ==> not Hackrf found  

hackrf_info
hackrf_info version: git-3344ea8
libhackrf version: git-3344ea8 (0.7)
**No HackRF boards found**.

And, do not ask me why , but  it was surprisingly solved , just connecting to windows, putting there in hackrf-mode  and from there flash the same binary with the bat .  (So windows recognised and fixed the problem )

I did not comment before , because that was other different experience ...

TQMatvey, 2022-05-10:
i tried from windows also :c

Brumi-2021, 2022-05-10:
Sorry, not so much new ideas...
But when you make my original described process , 
Just after dfu util command , if you send hackref_info command, you do not get : Found HackRF

TQMatvey, 2022-05-10:
i get Hackrf Not Found
[image]

TQMatvey, 2022-05-10:
@TheDaylightWalker after running your command (modified a bit cuz didnt work) nothing changed, dfu dissapeared and black screen and hackrf didnt appear
[image]
After manually rebooting to dfu the dfu-util --list returned this
[image]

TheDaylightWalker, 2022-05-11:
could you please try 
https://github.com/eried/portapack-mayhem/wiki/Won't-boot

Brumi-2021, 2022-05-11:
Hi @TQMatvey , 
I do not know if your device has built in the integrated battery , but if  yes , pls check that link 
https://github.com/greatscottgadgets/hackrf/issues/456#
Some other users that had problems with DFU , I think they solved disconnecting the internal battery   in that restoring operation.
Check and read all the link  comments, and try it (just in case , that it may help you ) 
Cheers

TQMatvey, 2022-05-11:
> could you please try https://github.com/eried/portapack-mayhem/wiki/Won't-boot

I already chedked, my device boots fine, hackrf mode is just broken

TQMatvey, 2022-05-12:
> Hi @TQMatvey , I do not know if your device has built in an integrated battery , but if yes , pls check that link, because big possibilities to fix it : [greatscottgadgets/hackrf#456](https://github.com/greatscottgadgets/hackrf/issues/456) Some other users that had problems with DFU , I think they solved disconnecting the internal battery in that restoring operation, (and then repeat all the DFU process that I mentioned you before in my first answer.) Check and read all the link comments, and try it (just in case , that it may help you ) . Cheers

Uff taking whope portapack h2 in metal case apart sounds bad..

Brumi-2021, 2022-05-12:
Well if you do not have also integrated speaker , taking care in the disassembly process  it should be possible. 
I would try , but taking time, and with special care .

Brumi-2021, 2022-05-13:
@TQMatvey any update or progress about it ?

TQMatvey, 2022-05-13:
Thanks everyone, today i took everything apart, detached everything, took pure hackrf board, flashed dfu, flashed hackrf_one.bat, put on portapack (no battery yet), flashed portapack, took portapack away, connected battery back, and other stuff back and everything works again, i am able to continue to work on firmware till another brick :)

closed 2022-05-13 by TQMatvey

Brumi-2021, 2022-05-13:
To @TQMatvey very happy to read that you manage to recover the brick 👍 . I also take note about it because very soon I will have a second similar clone as yours ,  H2 WM8731 with battery .

TQMatvey, 2022-05-15:
@Brumi-2021 could you accept my discord friend req?