portapack-mayhem/mayhem-firmware #642: Analog Audio - Using Stored Frequencies

infectedroot, 2022-05-11:
Whenever I'm listening to Analog Audio (via Receivers -> Audio), I've always thought it would be nice if I could select frequencies from a predetermined list to jump over to some of my favorite really quickly. Kind of how you do in the Looking Glass.

Additionally, it would be nice to also be able to save the frequency from the Analog Audio app as well.

My feature request aside... I wanted to say that I just got a HackRF and PortaPack H2 and am absolutely loving it. I'd eventually love to contribute to the project. However, it's been a while since I've programmed in C++. So, I'll stick to playing with the example in the docs (thanks, they're amazing!), and even try my hand at this enhancement.

So yeah - what does the rest of the community think?

GullCode, 2022-05-11:
I don't know for the community, but the Scanner (and or the Recon app I am making) is doing just what you are asking for.

ArjanOnwezen, 2022-05-11:
> I don't know for the community, but the Scanner (and or the Recon app I am making) is doing just what you are asking for.

fully agree @GullCode 
@infectedroot  Please check Scanner app or the new Recon app from @GullCode. Just join the discord server.

closed 2022-05-11 by ArjanOnwezen