portapack-mayhem/mayhem-firmware #645: Pocsag WFM in stead of NFM? White noise!

loewal, 2022-05-13:
**Describe the bug**
Only white noise after setting Pocsag to WFM

**To Reproduce**
Steps to reproduce the behavior:
1. goto Pocsag in Receive

**Expected behavior**
Sound and messages, but only white noise
And I think in the Netherlands it is NFM

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release: ok
* Nextt version of 13-05-2022

loewal, 2022-05-13:
Was already corrected in the comments

ArjanOnwezen, 2022-05-13:
What do you mean @loewal ?

ArjanOnwezen, 2022-05-13:
With latsest code pocsag uses NFM. And it works perfectly fine in The Netherlands. Hence I tried it before merging.

closed 2022-05-13 by ArjanOnwezen

ArjanOnwezen, 2022-05-13:
[image]

loewal, 2022-05-16:
Hello [ArjanOnwezen](https://github.com/ArjanOnwezen)
Sorry for opening again!

I am running SDRangel v7 beta3; device Portapack Hackrf H1
--> Pager Demodulator --> Pocsag

Ran for a few minutes and got really a load of messages; screenfull.

Than ran via the menu of the same device with firmware  n_220514:

1 or 2 lines over 5 minutes.

What could be wrong?

ArjanOnwezen, 2022-05-16:
Not sure. It could be a bug in autoswitching in Pocsag. Not sure to be honest.

ArjanOnwezen, 2022-05-16:
Please discuss in channel # pocsag app on Discord or comment in #478 . However WFM is not used.

tony1492, 2022-05-30:
Gee Since I upgraded to Version 1.5.3 I have no control over what baud rate to choose. The + side of it is now I can actually hear the sounds unlike in version 1.4. (I forgot) What should I do?

zigad, 2022-06-10:
> Gee Since I upgraded to Version 1.5.3 I have no control over what baud rate to choose. The + side of it is now I can actually hear the sounds unlike in version 1.4.3
> What should I do?

Since automatic baud detection Pocsag RX is useless, for me none of messages are decoded

tony1492, 2022-06-17:
> > Gee Since I upgraded to Version 1.5.3 I have no control over what baud rate to choose. The + side of it is now I can actually hear the sounds unlike in version 1.4.3
> > What should I do?
> 
> Since automatic baud detection Pocsag RX is useless, for me none of messages are decoded

That seems to be the case How do we get this fixed : ()