portapack-mayhem/mayhem-firmware #646: Not all settings are saved n_220514

Trufax, 2022-05-15:
**Describe the bug**
When i power cycle my Portapack H2 not all settings are saved. Specifically touchscreen (on/off) and time.
My PPH2 has an build on battery and the small CR1220 cell both attached. SD Card is available as well

**To Reproduce**
Set time and Touchscreen (on/off)
Powercycle the device
Most settings are stored (eg. turnoff screen after 5 sec) but Time and touchscreen settings are returned to default

**Expected behavior**
All settings should be stored

**Affected versions**
n_220514

ArjanOnwezen, 2022-05-15:
Touchscreen / time settings aren't saved on SD.
Please try older firmware version

jLynx, 2022-05-15:
Can anyone else reproduce this issue? Possibly too much data in the persistent storage, so it could be getting corrupted

TQMatvey, 2022-05-15:
Nothing from settings tab gets saved for me on latest nightly builds

jLynx, 2022-05-15:
How far back until it starts working?

TQMatvey, 2022-05-15:
I think all nighties after v1.5.1... I haven't used hacked much earlier but on 1.4.3 everything was perfect

jLynx, 2022-05-15:
If you could try all of them till you find the one that causes the issue that would be great. I can't reproduce this issue. So we will be relying on your information in order to fix it

TQMatvey, 2022-05-15:
> If you could try all of them till you find the one that causes the issue that would be great. I can't reproduce this issue. So we will be relying on your information in order to fix it

Ok will do now

ArjanOnwezen, 2022-05-15:
And @TQMatvey could you tell us wich chipset you're using? Same question for @Trufax

TQMatvey, 2022-05-15:
How to check?

TQMatvey, 2022-05-15:
Hi settings don't save even on versions that worked fine before...

TQMatvey, 2022-05-15:
Date and time also reset on 00:00:00 each reboot

jLynx, 2022-05-15:
Try remove your rtc battery and then insert it again. That clears any corrupted memory

TQMatvey, 2022-05-15:
Ah s#it I can't take my portapack apart now :( well will try some time later

ArjanOnwezen, 2022-05-15:
Hi @jLynx If it's related to settings and storage size it might be an issue for some devices that an extra field has been added in persistent_memory

uint32_t hardware_config;
	

This might be too much (memory) in some cases. 
Could you reuse playdead instead ?
So then size stays the same.

Also perhaps only use the first few bits from playdead (renamed to hardware_config) instead of do a cast to uint_8_t?

Not sure if it helps...

jLynx, 2022-05-15:
Don't think it's related to that as he said 1.5.1 use to work and the change was there already

ArjanOnwezen, 2022-05-15:
> Don't think it's related to that as he said 1.5.1 use to work and the change was there already

Clear..

TQMatvey, 2022-05-15:
At least for me, I can't confirm it worked on 1.5.1 100% I remember that it probably worked, but it's like 80/20

Trufax, 2022-05-15:
Hi, so i dont know how to identify the chipset version but i created a picture of the which might help to identify the version:
[image]

Also i flashed back to 1.5.1 and removed the CR battery but still Time is not saved. The rest of the settings seem to be all saved.

ArjanOnwezen, 2022-05-15:
Looks like this one
https://www.mmdvm.club/index.php/archives/197/

jLynx, 2022-05-21:
Is this issue still happening on 1.5.3?

Trufax, 2022-05-21:
Just flashed 1.5.3 and the Time is still not saved.

Brumi-2021, 2022-05-21:
@Trufax hi , but you installed back the CR1220 cell battery , not ?

Trufax, 2022-05-21:
With CR1220 cell.

ArjanOnwezen, 2022-05-21:
And other settings are saved, like time out period.

Brumi-2021, 2022-05-21:
This week I received apparently the same device than you , PCB 3.6 , but in your photo is not visible your audio codec Ic (but you can check in menu debug , peripherals , but just looking surround SMD's almost sure WM8731 same as mine) Anyway the first day , the clock  set up date , was lost everytime.  I already changed many fws , including jumbo and now Mayhem  . And I do not know from when, but recently it is working well now .

jLynx, 2022-05-21:
This may sound strange. But can you try flash the same firmware 4 times in a row? I remember someone a few months back saying they had to do that to get it working

Trufax, 2022-05-21:
flashed it 4x but still no save.

jLynx, 2022-05-21:
Ah, it was worth a shot

Brumi-2021, 2022-05-22:
Do you still have the unit opened outside of the metal case ? Can you reconfirm with a tester a good cell CR1220 voltage ? (I do not think that this will be the problem, but to discard ) . Other question , when you set up the clock , press the save . 2 or 3 min later without switching off , the clock is updating the time correctly in on conditions, no ?

ArjanOnwezen, 2022-05-27:
@Trufax this doesn't seem software related. Closing issue

closed 2022-05-27 by ArjanOnwezen