portapack-mayhem/mayhem-firmware #647: Problem upgrading the firmware

markotime1, 2022-05-17:
**What is happening?**
Describe here why you are unable to upgrade the firmware. Before describing your problems, **do the following**: 
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
Add photos and videos of your procedure.

jLynx, 2022-05-17:
Invalid issue

closed 2022-05-17 by jLynx

markotime1, 2022-05-17:
Upgrade yo 1.5  won't boot into mayhem. Downgrade to 1.4 all is OK. ???

jLynx, 2022-05-17:
Please read the wiki before creating issues https://github.com/eried/portapack-mayhem/wiki/Won't-boot

markotime1, 2022-05-17:
Tried all suggestions, including holding buttons.  No joy.  ....

jLynx, 2022-05-17:
Please edit your original issue to add all details including what portapack version you have. Also I assume you tried the latest nightly version?

As right now as it stands this is an invalid issue and it's not possible to help as you have provided no details

markotime1, 2022-05-18:
Thank you, I will do that.

markotime1, 2022-05-18:
Problem mysteriously vanished.now using nightly.  A question remains about
sensitivity, 2020 version hardware, aliexpress, amp on off shows up on
screen with noise level, but no real signals but for a few spurs. Whip
antenna.  Has yhere been discussion on this?  Thread,? Thanks so much for
your help

ArjanOnwezen, 2022-05-18:
> Problem mysteriously vanished.now using nightly. A question remains about sensitivity, 2020 version hardware, aliexpress, amp on off shows up on screen with noise level, but no real signals but for [...]

You discord channel. See link on main page