portapack-mayhem/mayhem-firmware #648: Problem upgrading the firmware

mtnbkr88, 2022-05-19:
For the v1.5.1 release, executing flash_portapack_mayhem.bat fails because it cannot find MSVCP120.dll. Previous releases included MSVCP120.dll and MSVCR120.dll in the firmware folder. I had to copy these from the v1.4.4 firmware folder to get the firmware upgrade to complete. Please include these two files in future firmware upgrade folders. Thanks.

ArjanOnwezen, 2022-05-19:
Use latest nightly

closed 2022-05-19 by ArjanOnwezen

markotime1, 2022-05-20:
Also, driver install defaults to AMD. Solution.. right click in the .inf file  and install. This allows not only visibility of hackrf  bur communication.  I needed to do both procedures to flash (nightly).

mtnbkr88, 2022-05-20:
I assume from the "Use latest nightly" comment that this has been recently fixed, but many of us prefer to only use stable releases and not be guinea pigs for what changed the night before. My suggestion is to fix the 1.5.1 zip firmware release and add back the missing two files. This might save others from wasting 30+ minutes of their lives they'll never get back trying to figure out why installing the latest vcredist files to get MSVCP120.dll doesn't fix the problem. I was lucky I had saved the v1.4.4 firmware distribution files to compare against to find the two missing files.

jLynx, 2022-05-20:
Hey well instead of complaining @mtnbkr88, how about you open a PR with those changes eh?

mtnbkr88, 2022-05-20:
I did open what I thought was a PR (#648) and @ArjanOnwezen gave what I thought was a snide comment and closed it. I'm trying to help here. Since you know there is a problem with the v1.5.1 zip firmware distribution. Fix it to save others some trouble.

ArjanOnwezen, 2022-05-20:
Hi @mtnbkr88, it's nothing personal, just telling you the problem has been resolved in latest nightly.
Either use latest nightly or wait until next release.
We tend to close tickets when it's resolved in a (nightly) release.

jLynx, 2022-05-21:
Fixed in https://github.com/eried/portapack-mayhem/pull/654