portapack-mayhem/mayhem-firmware #650: Problem upgrading the firmware

mtnbkr88, 2022-05-20:
I just installed the latest 1.5.3 firmware and found MSVCP120.dll and MSVCR120.dll are still missing from the mayhem_v1.5.3_FIRMWARE.zip file. Maybe whoever puts this file together should test it on a clean PC so they can see the errors a typical non-developer user might see. Please put these files back into the firmware distribution folder like they used to be in versions 1.4.4 and earlier. Just sayin'.
[MissingDLLs.zip](https://github.com/eried/portapack-mayhem/files/8744883/MissingDLLs.zip)
.

jLynx, 2022-05-21:
This isn't a PR, this is an issue. You should open a PR with the included missing files

jLynx, 2022-05-21:
I just did it as its a 2 second job. This will be available in the next nightly

closed 2022-05-21 by jLynx