portapack-mayhem/mayhem-firmware #656: View Map

TriniHam, 2022-05-22:
I think it would be nice to have an option for us to navigate the map. Maybe we enter coordinates and the corresponding data will be displayed  🤷‍♀️. This will be typically useful for those like myself when working on the world_map or a portion of it.

ArjanOnwezen, 2022-05-27:
@TriniHam That's the case right now, right. Map is based on coordinates from AIS or Radiosonelde position.

TriniHam, 2022-05-30:
> @TriniHam That's the case right now, right. Map is based on coordinates from AIS or Radiosonelde position.

I mean like an actual enter coordinates and view app. Radiosonelde  and those apps does only track that frequency correct?. Also generate_world_map.bin.py isn't working for me in VSC is there another compiler I can use to run this script? Thanks

ArjanOnwezen, 2022-05-30:
But why would you want to enter a manual coordinate? Just use OpenStreetMap for an offline map.

TriniHam, 2022-05-30:
> But why would you want to enter a manual coordinate? Just use OpenStreetMap for an offline map.

I would like to view only part of the map at a certain zoom that's why I want to be able to enter the coordinates. And I can't put the entire world on one zoom level, it will be HUGE.

TriniHam, 2022-05-30:
> > But why would you want to enter a manual coordinate? Just use OpenStreetMap for an offline map.
> 
> I would like to view only part of the map at a certain zoom that's why I want to be able to enter the coordinates. And I can't put the entire world on one zoom level, it will be HUGE. Also after creating the map the only way to test it in portapack is to wait for an aircraft to file over, which doesn't always be convenient

eried, 2022-05-31:
Since your request is quite specific, In this case I think the easiest would be to [clone ](https://github.com/eried/portapack-mayhem/wiki/Create-a-simple-app)one of the apps, but that will challenge you :) maybe you can make a new utility for everyone called "mapviewer". I cannot actually imagine anyone prefering to browse the map in the device vs openstreetmap lol tho

TriniHam, 2022-06-01:
> Since your request is quite specific, In this case I think the easiest would be to [clone ](https://github.com/eried/portapack-mayhem/wiki/Create-a-simple-app)one of the apps, but that will [...]

Thanks, will try

ArjanOnwezen, 2022-06-01:
> > Since your request is quite specific, In this case I think the easiest would be to [clone ](https://github.com/eried/portapack-mayhem/wiki/Create-a-simple-app)one of the apps, but that will challenge you :) maybe you can make a new utility for everyone called "mapviewer". I cannot actually imagine anyone prefering to browse the map in the device vs openstreetmap lol tho
> 
> Thanks, will try . Is there any other way to run the code without compiling then flashing into portapack?

Nope

gullradriel, 2023-07-18:
Solved by https://github.com/eried/portapack-mayhem/pull/1172

closed 2023-07-18 by gullradriel