portapack-mayhem/mayhem-firmware #657: how to upgrade clpd

deboi238, 2022-05-23:
im confuse how to update the clpd, does anyone have any tutorial how to update the firmware?

jLynx, 2022-05-23:
Before creating this issue, **do the following**: 
* Read the Wiki on booting: https://github.com/eried/portapack-mayhem/wiki/Won't-boot
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

closed 2022-05-23 by jLynx

deboi238, 2022-05-24:
> Before creating this issue, **do the following**:
>
> * Read the Wiki on booting: https://github.com/eried/portapack-mayhem/wiki/Won't-boot
> * Read: [...]

so when I tried to upgrade it, it have a error state and this error code I have _wfopen: 80070002

jLynx, 2022-05-24:
It seems like a problem with the flashing software. I would recommend trying another computer and/or usb cable.

deboi238, 2022-05-24:
> It seems like a problem with the flashing software. I would recommend trying another computer and/or usb cable.
i already tried 2 different cable but still wont work though

deboi238, 2022-05-24:
> It seems like a problem with the flashing software. I would recommend trying another computer and/or usb cable.

so how can I upgrade the clpd? since I have to use 1.4.3 version that someone provided for the clpd

jLynx, 2022-05-24:
When you say clpd I assume you mean firmware upgrade. Your issue is invalid which is why I closed it. You have not followed the correct procedure for creating one. If you are the same person I am talking to on discord, there was no need to create this issue

deboi238, 2022-05-24:
> When you say clpd I assume you mean firmware upgrade. Your issue is invalid which is why I closed it. You have not followed the correct procedure for creating one. If you are the same person I am talking to on discord, there was no need to create this issue

I have followed the exact same thing on wiki and on video, all I have is black screen and error led and I don't have discord