portapack-mayhem/mayhem-firmware #658: Standby frequency

Dadanos, 2022-05-24:
Hello everyone,

I was wondering if is possible to implement on audio receive app (or to more) a "standby frequency" and to switch to this by pressing a "frequency transfer button" like all aircraft transceivers.
This way is possible to monitor almost instantly 2 frequencies. 
Such feature will be extremely useful for monitoring air band and following the line of communications e.g. aircraft switching from approach frequency to tower frequency.
It could be a 0-1 setting like the amp function on the blue strip next to step size
Unfortunately i am not much of a coder myself to know the level of difficulty to implement such feature.
I have attached a photo of a common transceiver for reference.

P.S. Sorry for my English, not a native speaker. 

Best regards.
   
[image]

eried, 2022-05-24:
It would be like a "swap" from the list of frequencies you can load for example, but how do you expect the interface to work? I mean what is the user interaction required to swap in the ideal scenario of your idea?

Dadanos, 2022-05-24:
Exactly like "swap"! For example by making selectable the "Stby + number" of the attached modified screenshot (like the step) and pressing the center button or by turning the rotary encoder (just a suggestion) personally i don't mind as long as it works!
Or it could read the frequencies from a txt file of the frequency manager and just go to the next one.
Or it could be just a <-> button instead of the "Stby" text and by pressing it would swap the frequencies.
I really don't know what is the easiest way to add something like this, i am just trying to provide inspiration.
 [image]